tuya-cloudcutter/tuya-cloudcutter #477: Cloudcutter not connecting to AP

Atsukio, 2023-09-10:
So i flashed a couple of the tuya devices with cloudcutter and it's been working great, but only this specific 3 Gang light switch one isn't connecting.

It creates a "SmartLife-C531" AP, i've even checked and even my phone sees the SmartLife AP, but cloudcutter it keeps scanning for SmartLife APs.

`shell
Selected Device Slug: tuya-generic-ds-103-3-gang-light-switch
Selected Profile: slg-bwkj-k3-1.1.0-sdk-1.0.2-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.............................................................................................................................................................................................................

`

Cossid, 2023-09-10:
Try rebooting the device running CloudCutter and optionally running with the `-r` parameter.

Atsukio, 2023-09-10:
> Try rebooting the device running CloudCutter and optionally running with the `-r` parameter.

Oh nice `-r` solved it thanks.

closed 2023-09-10 by Atsukio