tuya-cloudcutter/tuya-cloudcutter #478: New Device - realme Wi-Fi 6A Smart Plug

dasshubham762, 2023-09-10:
Link: [realme Wi-Fi 6A Smart Plug](https://www.flipkart.com/order_details?order_id=OD222880335874429000&item_id=22288033587442900&unit_id=22288033587442900000)

Dump: [dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12568525/dump.zip)

Cossid, 2023-09-10:
The dump is valid, but this build doesn't match existing patterns, so I'll need a bit of time to adapt haxomatic to support this device.

Cossid, 2023-09-10:
@dasshubham762 This device doesn't appear to have a device configuration, do you know which Beken BK7231T module this has?  Do you know which pin the physical button maps to?

Cossid, 2023-09-11:
@dasshubham762 I can't test this myself because I don't know what pin button is bound do, and it doesn't appear to be one on the board I have, so please extract this file into your `device-profiles` directory and use the argument `-p realme` and see if it works for you.
[realme.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12571150/realme.zip)

dasshubham762, 2023-09-11:
> @dasshubham762 This device doesn't appear to have a device configuration, do you know which Beken BK7231T module this has? Do you know which pin the physical button maps to?

I have seen that this device has Tuya WB2S module, which I have watched on a tear down video. But I can't say which button maps to the physical button. It's pretty difficult to open the device.

dasshubham762, 2023-09-11:
> @dasshubham762 I can't test this myself because I don't know what pin button is bound do, and it doesn't appear to be one on the board I have, so please extract this file into your `device-profiles` [...]

Okay. I'll test an let you know.

dasshubham762, 2023-09-11:
@Cossid These are the GPIO maps listed below (Figured out with ESPHome Kickstart):

P6/PWM0 = Button
P7/PWM1 = B LED
P24/PWM4 = Relay
P26/PWM5 = R LED

dasshubham762, 2023-09-11:
Yes. The profile worked correctly.

Cossid, 2023-09-11:
Thanks for confirming, tools have been updated to support that new binary type and a device profile has been added as `realme RMH2009 6A Smart Plug`

closed 2023-09-11 by Cossid