tuya-cloudcutter/tuya-cloudcutter #480: Tuya version 1.5.21 missing?

jenshenk, 2023-09-10:
Hi,
Just got a new GU10 RGBW LED spot from MOES WB-TD5-RWW-GU10-MS. Read it used to run tuya version 1.3.21 but mine is 1.5.21 which is not in the list of tuya cloudcutter. Tried a couple of other versions without success. Are there plans to add support for this version or can I use another version from the list? Just keeping on trying would take a lot of time and potentially not lead to success. Thanks for any tips or infos!

Cossid, 2023-09-10:
1.5.21 is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware list, and is not exploitable by CloudCutter.  Serial flashing will be your only option.

closed 2023-09-10 by Cossid

jenshenk, 2023-09-10:
Thanks for the super quick answer! Too bad but guess my luck had to end at some point, flashed 20 tuya devices with cloudcutter and this is the second in a row where I need to resort to serial flashing.

mandarons, 2023-10-20:
Any BKM on how to flash ESPHome firmware using serial?

Cossid, 2023-10-20:
https://docs.libretiny.eu/docs/flashing/esphome/