tuya-cloudcutter/tuya-cloudcutter #481: Flashed Smart Plug (1.3.5 oem_bk7231s_dltj) with ESPHome but pins don't activate the relay

4rianton, 2023-09-12:
Bought a generic smart plug from Ali and it had firmware 1.3.5. I started Tuya Cloudcutter and chose the only device that appeared for 1.3.5 firmware (oem_bk7231s_dltj). The flash was successful, and I installed ESPHome kickstart firmware on it. 

Now, in ESPHome, when I try to figure out the pin mappings by going through the pins one by one and setting their output high and low to see if the plug reacts... but even after going through all the pin mappings, the plug doesn't turn on. It worked fine when it had Tuya on it, so I can rule out a hardware issue. 

What can I try to get the relay to work? Is it because I just chose "oem_bk7231s_dltj" even though it's likely not accurate since dltj is probably not a plug? 

Looking forward to some pointers. :)

kuba2k2, 2023-09-12:
Download [ltchiptool](https://github.com/libretiny-eu/ltchiptool) (the Windows GUI version), go to UPK2ESPHome tab, and press "Grab from Kickstart" button. This will allow you to read the Tuya configuration that is stored on the device (it's a generic plug firmware, so 99% chance that your YAML config can be generated automatically this way).

4rianton, 2023-09-12:
I use a Mac but can organise a Windows machine soon. However, do you reckon that getting a YAML will change anything even though changing pin-mode for every pin available didn't turn on the relay? 

> Download [ltchiptool](https://github.com/libretiny-eu/ltchiptool) (the Windows GUI version), go to UPK2ESPHome tab, and press "Grab from Kickstart" button. This will allow you to read the Tuya [...]

kuba2k2, 2023-09-12:
Yes. It also depends on which Kickstart version you used - the v23.07.13 one had a bug in pin mapping, which made most pins unresponsive.

4rianton, 2023-09-12:
How interesting. I happen to have 23.07.13 kickstart. Let me update this first and report back.

4rianton, 2023-09-12:
That did it! Updating kickstart did it. Thank you for that pointer. Cheers!

closed 2023-09-12 by 4rianton