tuya-cloudcutter/tuya-cloudcutter #482: New Profile for Temperature Sensor

Francesco-Sch, 2023-09-12:
Hey, 
I pulled the attached firmware from this smart life temperature and humidity sensor over serial with bk7321tool: 
https://www.amazon.de/dp/B0B9Y4D4PR?psc=1&ref=ppx_yo2ov_dt_b_product_details

I think it must be the same as this. At least the inside components are the same as well as the Main Module number:
https://de.aliexpress.com/item/1005004755749389.html?spm=a2g0o.order_list.order_list_main.45.79ef5c5fQDqyAe&gatewayAdapt=glo2deu

I tried to build a new profile myself but I could not get a token, because my version of Smart Life is too new. 
Could mabye somebody help me with that? I also need it kind of urgently so any help would be really appreciated. :) 

Thanks and best
Francesco

---

[eMylo_th08---Temperature-Humidity-Sensor.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12586970/eMylo_th08---Temperature-Humidity-Sensor.bin.zip)

Francesco-Sch, 2023-09-12:
If this works we could also close #473

Francesco-Sch, 2023-09-12:
Or if somebody could tell me how to acquire the token with the smart life app approach, I would be really thankful. :)

kuba2k2, 2023-09-12:
Is that a dump from the device you've tried to flash in #473?

Francesco-Sch, 2023-09-12:
Nope, that is a different one. :)

kuba2k2, 2023-09-12:
Okay. Why do you need the profile urgently if you've already managed to dump the firmware? (and you can flash ESPHome via UART too)

Francesco-Sch, 2023-09-12:
I am working on an art project for my arts university, where I need to flash a couple of those temperature sensor to control them for an installation. 

And I thought it might be easier if I just need to flash one via serial and I can flash the rest with tuya-cloudcutter so I do not need to open everyone of them. :) 

In addition I also wanted to generate a YAML for ESPHome from that profile with something like [UPK2ESPHome](https://upk.libretiny.eu/) so I already have all the pins and controls in that YAML. Does this make sense? Or can I generate that also from the firmware directly? :)

Francesco-Sch, 2023-09-12:
By the way this is the log when I tried to build the profile myself: 

```
tuya-cloudcutter/profile-building [main] » python3 build_profile.py eMylo_th08---Temperature-Humidity-Sensor.bin
[+] Processing file='eMylo_th08---Temperature-Humidity-Sensor.bin' as eMylo_th08---Temperature-Humidity-Sensor
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xea20]
		extracted to /Users/fscheffczyk/DevProjects/tuya-cloudcutter/profile-building/eMylo_th08---Temperature-Humidity-Sensor
	0x129f0a: app - [encoding_algorithm=NONE, size=0xdfa20]
		extracted to /Users/fscheffczyk/DevProjects/tuya-cloudcutter/profile-building/eMylo_th08---Temperature-Humidity-Sensor
Storage partition:
	0x1ee000: 68 KiB - 11 keys
	- 'gw_bi'
	- 'gw_di'
	- 'tls_ca_cnt'
	- 'wf_start_md'
	- 'gw_wsm'
	- 'timer_arr'
	- 'gw_ai'
	- 'is_stride'
	- 'record_head'
	- 'baud_cfg'
	- 'em_sys_env'
		extracted all keys to /Users/fscheffczyk/DevProjects/tuya-cloudcutter/profile-building/eMylo_th08---Temperature-Humidity-Sensor/eMylo_th08---Temperature-Humidity-Sensor_storage.json
App code `user_param_key`:
	- not found!
[+] Searching for known exploit patterns
[+] Matched pattern for BK7231N version SDK 2.3.1, payload type ssid
[+] Searching for ssid payload address
[+] ssid payload address gadget (THUMB): 0xC64B5
[+] Searching for finish address
[+] Finish address gadget (THUMB): 0x1B52F
[+] uuid: f0aff6ba4254392f
[+] auth_key: u7kToY6e0rYsq9393BNeJfJAGmTXcg5V
[+] ap_ssid: SmartLife
[+] swv: 2.1.0
[+] bv: 40.00
[+] firmware key: keyr5qhaxgstx9ys
[+] product key: xlk3mtpjo1zkmdvh
[!] No gw_di, No version or key stored, manual lookup required
[+] SDK: 2.3.1
[+] Device class: bk7231n_lowpower_common_ty
[!] No token provided.
[!] On any device on the same network as you're device running this script, please log into the Smart Life app ('Try as Guest' works fine if you do not already have an account)
[!]  Note: no real device should currently be in pairing mode, or the instructions below may not match.
[!]  - Start the add device procedure (hit '+' in the upper-right and select 'Add Device')
[!]  - Under 'Add Manually' select 'Socket (Wi-Fi)'
[!]  - Enter your network credentials as instructed
[!]  - Do not follow any of the instructions about putting a device into pairing mode, instead select 'next' until it asks the status of the indicator and select 'Blink Slowly'
[!]  - Select 'Go to Connect', then in your wifi selection screen, hit the back button to return to Smart Life.
[!] A new token should be sent to your network, and this script will continue.
[!] Note: this will join an unresponsive device to your account.  You can safely delete it afterwards.
[+] Waiting for multicast token from app...
^C[!] Cancelled waiting for token.
[!] Error receiving new token.

```

Maybe this helps. :)

Cossid, 2023-09-12:
Added as `eMylo CSAC905 Temperature and Humidity Sensor Display` (may take a few minutes to process).

You're not easily going to be able to control this display, as it is a TuyaMCU device.

closed 2023-09-12 by Cossid

Francesco-Sch, 2023-09-12:
Thanks @Cossid, really appreciate the help. :)

What could become the problem with controlling the screens of TuyaMCU devices?

Cossid, 2023-09-12:
I mean you won't be able to customize it, it's controlled by the TuyaMCU, and since the temperature sensor is also behind the MCU and not GPIO, you can't really change the values much.  You'll mostly be limited to what they already support.

Cossid, 2024-11-07:
This must have been processed before we had proper checks for patched binaries.  This one actually is patched (and on the known patched list since), so the profile for it will be removed, since it cannot work.