tuya-cloudcutter/tuya-cloudcutter #485: GR-W-105 Water Shutoff Valve 1.15 - profile selected did not result in a successful exploit

ntsmkfob, 2023-09-19:
The model I have may be a clone of the GR-W-105 valve. It uses a WB3S with Main and MCU modules both 1.1.5. Is there anything I can do to narrow down the reason for failure? 

I also have another unit with both modules at 1.18. This fails when I choose the module for a 1.1.8 Plug. Can I do anything to uncover what changes have been made?

Thanks

Cossid, 2023-09-19:
> The model I have may be a clone of the GR-W-105 valve. It uses a WB3S with Main and MCU modules both 1.1.5. Is there anything I can do to narrow down the reason for failure?

There are several 1.1.5 firmwares, including for both chip types.  Can you confirm which module is inside?  It would help narrow it down.

> I also have another unit with both modules at 1.18. This fails when I choose the module for a 1.1.8 Plug. Can I do anything to uncover what changes have been made?

Again, 1.1.8 also has multiple firmwares, including multiple chips.  Knowing the module again would help narrow it down.

Instead of picking devices out, choose the option of by firmware version, and select each one and see if one works.  If any of them makes it past "profile selected did not result in a successful exploit" mark it down and continue using it, or it gets confusing later.

Cossid, 2023-09-19:
Sorry missed your WB3S part on the first one.  The `Tuya Generic GR-W-105 Water Shutoff Valve v1.1.5` *should be* correct for that device.

ntsmkfob, 2023-09-20:
Thanks. As I used a RPI I had already set up, I'll start from scratch in case that had an impact on the result.

ntsmkfob, 2023-09-23:
No change after fresh install of RaspiOS Lite, docker and cloudcutter and run with the GR-W-105 Smart Valve with 1.1.5 firmware. Is there any debugging I can do to help?

Cossid, 2023-09-23:
Please post complete run logs of the cutting process, along with [proof of what firmware is currently running on the device](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has).

ntsmkfob, 2023-09-23:
Main output from  run:

Selected Device Slug: tuya-generic-gr-w-105-water-shutoff-valve-v1.1.5
Selected Profile: oem-bk7231s-rnd-zero-dctrl-switch-1.1.5-sdk-1.0.6-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

Scanning for open Tuya SmartLife AP
................
Found access point name: "SmartLife-BF8B", trying to connect...
Device 'wlan0' successfully activated with '0c0ea6d3-6b33-44b0-9701-04512066dfcf'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.

Scanning for open Tuya SmartLife AP
................
Found access point name: "SmartLife-BF8B", trying to connect...
Device 'wlan0' successfully activated with '0c0ea6d3-6b33-44b0-9701-04512066dfcf'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.

Cossid, 2023-09-23:
Now please verify what version firmware your device is actually running, you can take a screenshot from the Smart Life app, you can find in in the instructions above.

ntsmkfob, 2023-09-23:
My previous msgs crossed with yours. I've attached a grab of the Device details and the output from the run.

[image]
[RunLog.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12706700/RunLog.txt)

Cossid, 2023-09-23:
> I notice the Profile is bk7231s, where the firmware chosen for KickStart is bk7231t - is that an issue?

No, that is not an issue, bk7231s and bk7231t are the same thing.

At this point, this can only mean your device has a different firmware.  The only ways forward are to try by version and see if the `1.1.5 - BK7231N / oem-bk7231n-control-switch-1.1.5-sdk-2.3.3-40.00` profile works, and if not, verify which module is actually inside and probably get a serial dump.

Cossid, 2023-10-18:
Closing as stale.  If you can provide a dump, please attach it here and re-open this issue.

closed 2023-10-18 by Cossid