tuya-cloudcutter/tuya-cloudcutter #486: New device: Tuya water leak sensor

TriggrHappy, 2023-09-24:
Device information:
https://aliexpress.com/item/1005004384160139.html

Flash dump:
[dump_tuya_water_leak_original firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12709458/dump_tuya_water_leak_original.firmware.zip)

Exploitable but none of the existing profiles worked for me.

Cossid, 2023-09-24:
Is that the correct product link?  I only see zigbee on that page.

Cossid, 2023-09-24:
Added as `Tuya Generic 898WT Water Leak Sensor` based on the name in the firmware.

closed 2023-09-24 by Cossid

TriggrHappy, 2023-09-24:
> Is that the correct product link? I only see zigbee on that page.

Sorry about that c&p mistake. But I see you managed to find the wifi version - that was fast :)