tuya-cloudcutter/tuya-cloudcutter #491: Geeni gn-ow102-103-oudoor-dual-plug will not install

billyjoebob999, 2023-09-30:
I'm not able to flash Geeni gn-ow102-103-oudoor-dual-plug. I using the profiles under manufacture but it comes back saying to can't exploit. I noticed that the Tuya app shows my device being on firmware/MCU version 1.0.2 so I tried all the 1.0.2 versions (see below) and it still fails. Not really sure why. Any suggest what might be going on?

`Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_strip_key_ir_mk
   1.0.2 - BK7231T / cuco_sw5_bk_common
   1.0.2 - BK7231T / oem_bk7231s_2_plug_globe_botai
   1.0.2 - BK7231T / oem_bk7231s_control_switch
   1.0.2 - BK7231T / oem_bk7231s_dltj_test
   1.0.2 - BK7231T / oem_bk7231s_light_mix4to5_db_rmd
   1.0.2 - BK7231T / oem_bk7231s_rnd_zero_dctrl_switch
 > 1.0.2 - BK7231T / oem_bk7231s_strip_key_ir_mk
   1.0.3 - BK7231N / oem_bk7231n_curtain_onoff
   1.0.3 - BK7231T / oem_bk7231s_control_switch
   1.0.3 - BK7231T / oem_bk7231s_dltj
   1.0.3 - BK7231T / oem_bk7231s_light_ty_oldDp
   1.0.4 - BK7231N / oem_bk7231n_strip_light4_FEIT
   1.0.4 - BK7231N / th03_temp_hum_sensor

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
   OpenBeken-v1.17.245_bk7231t.ug.bin

Selected Device Slug: iwiik-iwwctlrgb-rgbct-led-light-strip
Selected Profile: oem-bk7231s-strip-key-ir-mk-1.0.2-sdk-2.0.0-30.06
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
..
Found access point name: "Geeni-F1F1", trying to connect...
Device 'wlan0' successfully activated with 'ab953427-9e01-4593-8e27-50d88329fe03'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/cx7Y8QJkHBfT.deviceconfig
Saved device config in /work/configured-devices/cx7Y8QJkHBfT.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.
Found access point name: "Geeni-F1F1", trying to connect...
Device 'wlan0' successfully activated with '5b6e8b4b-0de8-4599-9d7e-ba21c745a589'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================`

Cossid, 2023-09-30:
> Tuya app shows my device being on firmware/MCU version 1.0.2 so I tried all the 1.0.2 versions (see below) and it still fails.

Your log only shows you trying on, specifically a strip, which is for sure not the right one.  You could try some of the ones that are more likely like any of the following:
```
1.0.2 - BK7231T / oem_bk7231s_2_plug_globe_botai
1.0.2 - BK7231T / oem_bk7231s_control_switch
1.0.2 - BK7231T / oem_bk7231s_dltj_test
1.0.2 - BK7231T / oem_bk7231s_rnd_zero_dctrl_switch
```
as they are all plug-related, but it looks like the one with 1.0.7 used `1.0.7 - BK7231T / oem_bk7231s_rnd_switch`, which means there may just not be a compatible profile for your device yet, which would mean a serial dump may be required to properly support it.

billyjoebob999, 2023-09-30:
Thanks for relying. You are correct, i only showed one log but I did try the ones you suggested plus the other 1.0.2 - BK7231T profiles. They all resulted the same so didn't post the logs for each since as it seemed un-necessary. I think you are right that it may not be compatible yet.

billyjoebob999, 2023-09-30:
Just following on this. After opening the device I found that it uses the TYWE3S chip. From the information I found this appears to be a ESP8266. Long story short I ended up using Tuya Convert to flash a ESPHome firmware on to the device.

closed 2023-09-30 by billyjoebob999