tuya-cloudcutter/tuya-cloudcutter #495: No profiles for firmware 1.3.3

apbarratt, 2023-10-02:
Hi guys, I have a TCP Smart LED strip (RGBW).  The chip labelling is WB3S, which I understand means it's a BK7231T.
The Smart Life app tells me the firmware is 1.3.3, however I cannot see any profiles listed with this version.

There's a number of 1.3.2 profiles, then a 1.3.21, and then it jumps to 1.3.5, having skipped 1.3.3 and 1.3.4 altogether.

Is there anything I can provide to help with this, or am I just out of luck for now :(

Cossid, 2023-10-02:
It would require a firmware dump to add support.  You would either need to dump via serial or LightLeak if you have a spare dev board and an android phone (and a bit of luck).

apbarratt, 2023-10-02:
Ah, I had feared as much.  Unfortunately I've not been very successful with trying to solder onto the contacts on this board so I don't think I'm going to have a lot of luck :(

apbarratt, 2023-10-02:
Hmmm, not off to a brilliant start.  I've built the custom project using the 8266 build to stick onto a NodeMCU thingy, but it doesn't seem to be able to connect to it over USB.  Perhaps I'm doing something wrong :/

apbarratt, 2023-10-03:
[dump-1.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12794637/dump-1.zip)
Here you go guys.  I don't know if this has actually worked successfully, the app didn't seem to think so, but there was stuff in the folder so here it is.

apbarratt, 2023-10-03:
Righto, a few more attempts and here's a complete dump :) 
[dump-3.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/12795445/dump-3.zip)

apbarratt, 2023-10-03:
This is the model that dump is from:
https://www.screwfix.com/p/tcp-5m-led-smart-rgbw-striplight-17w-1500lm/576pk

Cossid, 2023-10-03:
Added as `TCP SMAWTLRGBWIP205MUK 5M 17W RGBW LED Strip`, may take a few minutes to process and show in the list.

closed 2023-10-03 by Cossid