tuya-cloudcutter/tuya-cloudcutter #497: ssl error

kurtdefeyter, 2023-10-09:
please help the noob,
This all works fine up until , what i presume is the last phase , then getting a ssl error , anyone can point me in a direction how to solv this

Building cloudcutter docker image
[+] Building 3.6s (13/13) FINISHED                               docker:default
 => [internal] load .dockerignore                                          0.5s
 => => transferring context: 2B                                            0.0s
 => [internal] load build definition from Dockerfile                       0.5s
 => => transferring dockerfile: 460B                                       0.4s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-bus  2.2s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:83  0.0s
 => [internal] load build context                                          0.1s
 => => transferring context: 3.02kB                                        0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                   0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipe  0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                 0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                  0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:cfe4bb85910528b04f56aa9ee2de2c3a4f5fad044a44a  0.0s
 => => naming to docker.io/library/cloudcutter                             0.0s
Successfully built docker image
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 0.0.2 - BK7231N / oem_bk7231n_light_st                               [?] Select the firmware version and name: 0.0.3 - BK7231N / oem_bk7231n_light_st                               [?] Select the firmware version and name: 0.0.6 - BK7231N / oem_bk7231n_strip_43                               [?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_ceiling_                               [?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_dltj_plu                               [?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_garage_d                               [?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_jiabitai                               [?] Select the firmware version and name: 1.0.0 - BK7231T / HYS_03_018_bk7231t_t                               [?] Select the firmware version and name: 1.0.0 - BK7231T / oem_bk7231s_dltj_1pl                               [?] Select the firmware version and name: 1.0.1 - BK7231N / bk_light_rgbcw_vpylf                               [?] Select the firmware version and name: 1.0.1 - BK7231N / oem_bk7231n_scene_sw                               [?] Select the firmware version and name: 1.0.1 - BK7231N / oem_bk7231n_strip_ff                               [?] Select the firmware version and name: 1.0.1 - BK7231T / HYS_03_010_RGB_soq7i                               [?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_1_plug_a                               [?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_light_mi                               [?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_ty_ffc_n                               [?] Select the firmware version and name: 1.0.10 - BK7231N / bk7231n_lowpower_co                               [?] Select the firmware version and name: 1.0.10 - BK7231N / oem_bk7231n_gate_ma                               [?] Select the firmware version and name: 1.0.10 - BK7231N / oem_bk7231n_light_p                               [?] Select the firmware version and name: 1.0.10 - BK7231T / oem_bk7231s_light_t                               [?] Select the firmware version and name: 1.0.14 - BK7231N / oem_bk7231n_slide_s                               [?] Select the firmware version and name: 1.0.15 - BK7231N / oem_bk7231n_slide_s                               [?] Select the firmware version and name: 1.0.18 - BK7231N / oem_bk7231n_slide_s                               [?] Select the firmware version and name: 1.0.2 - BK7231N / bk7231n_common_user_                               [?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_dctrl_sw                               [?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_strip_sl                               [?] Select the firmware version and name: 1.0.2 - BK7231T / HYS_03_010_RGB_xhsx0                               [?] Select the firmware version and name: 1.0.2 - BK7231T / bk7231t_common_user_                               [?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_2_plug_g                               [?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_control_                               [?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_dltj_tes                               [?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_light_mi                               [?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_rnd_zero                               [?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_strip_ke                               [?] Select the firmware version and name: 1.0.3 - BK7231N / oem_bk7231n_curtain_                               [?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_control_                               [?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_light_ty                               [?] Select the firmware version and name: 1.0.4 - BK7231N / oem_bk7231n_strip_li                               [?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_light4_4                               [?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_light_st                               [?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_ty_ffc_n                               [?] Select the firmware version and name: 1.0.5 - BK7231N / oem_bk7231n_dctrl_sw                               [?] Select the firmware version and name: 1.0.5 - BK7231N / oem_bk7231n_light_at                               [?] Select the firmware version and name: 1.0.6 - BK7231N / oem_bk7231n_strip_rg                               [?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_dltj_2sw                               [?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_fan_cont                               [?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_light_db                               [?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_rnd_swit                               [?] Select the firmware version and name: 1.0.6 - BK7231T / realme_BK7231T_plug_                               [?] Select the firmware version and name: 1.0.7 - BK7231T / oem_bk7231s_rnd_swit                               [?] Select the firmware version and name: 1.0.7 - BK7231T / oem_bk7231s_rnd_zero                               [?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_light2_p                               [?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_light_db                               [?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_rnd_swit                               [?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_ty_ffc_d                               [?] Select the firmware version and name: 1.0.9 - BK7231N / BL_RGBCW_LIGHT_12KEY                               [?] Select the firmware version and name: 1.0.9 - BK7231T / oem_bk7231s_ty_ffc_d                               [?] Select the firmware version and name: 1.1.0 - BK7231N / oem_bk7231n_curtain_                               [?] Select the [... cut ...]
SSL Error on 15 ('10.42.42.36', 64852): [SSL: NO_SHARED_CIPHER] no shared cipher                                (_ssl.c:1129)
SSL Error on 15 ('10.42.42.36', 64853): [SSL: NO_SHARED_CIPHER] no shared cipher                                (_ssl.c:1129)
SSL Error on 15 ('10.42.42.36', 64854): [SSL: NO_SHARED_CIPHER] no shared cipher                                (_ssl.c:1129)
SSL Error on 15 ('10.42.42.36', 64855): [SSL: WRONG_SSL_VERSION] wrong ssl versi                               on (_ssl.c:1129)

many thanks

kurtdefeyter, 2023-10-09:
update : this update is that the error is the result of connecting the iphone to the ap ,
this following this doc
1. After your device gets stuck at the AP-ENABLED step - Just leave cloudcutter at the AP-ENABLED step (do not stop it, do not CTRL-C out if it - just leave it as it)
2. On your phone remove the location permission for the Tuya app
3. Start the Tuya app on your mobile phone
4. Manually add a new device selecting the device type you have
5. Put your device back into fast blinking pairing mode
6. In the Tuya app add your device and follow the steps (confirm that is fast blinking etc)
7. When it gets to the WIFI step - add cloudcutterflash as the WiFi name and abcdabcd as the password
8. Let the Tuya app try and add the device. When the device is added the fast blinking (pairing mode) stops
9. Go to your cloudcutter window and watch the logs - within a few minutes the update process will resume and the firmware will be successfully deployed to the device
  however the result is without connecting it thus get stuck on 
Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:9e:81:77 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

and there its waiting

Cossid, 2023-10-09:
I'd suggest using slow-blink mode instead then, as according to your log, your phone connected, but the device didn't.  Others have had more success using slow blink mode, entering the cloud cutter wifi credentials, then joining the `A-` AP to send those credentials.

kurtdefeyter, 2023-10-09:
many thanks , however i was able to flash the esp home firmware now the device is dead no more reaction our lights i am afraid id did dy
kurt@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh
Building cloudcutter docker image
[+] Building 4.3s (13/13) FINISHED                               docker:default
 => [internal] load .dockerignore                                          0.1s
 => => transferring context: 2B                                            0.0s
 => [internal] load build definition from Dockerfile                       0.1s
 => => transferring dockerfile: 460B                                       0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-bus  2.4s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:83  0.0s
 => [internal] load build context                                          0.5s
 => => transferring context: 3.02kB                                        0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                   0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipe  0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                 0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                  0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:cfe4bb85910528b04f56aa9ee2de2c3a4f5fad044a44a  0.0s
 => => naming to docker.io/library/cloudcutter                             0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 0.0.2 - BK7231N / oem_bk7231n_light_st[?] Select the firmware version and name: 0.0.3 - BK7231N / oem_bk7231n_light_st[?] Select the firmware version and name: 0.0.6 - BK7231N / oem_bk7231n_strip_43[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_ceiling_[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_dltj_plu[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_garage_d[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_jiabitai[?] Select the firmware version and name: 1.0.0 - BK7231T / HYS_03_018_bk7231t_t[?] Select the firmware version and name: 1.0.0 - BK7231T / oem_bk7231s_dltj_1pl[?] Select the firmware version and name: 1.0.1 - BK7231N / bk_light_rgbcw_vpylf[?] Select the firmware version and name: 1.0.1 - BK7231N / oem_bk7231n_scene_sw[?] Select the firmware version and name: 1.0.1 - BK7231N / oem_bk7231n_strip_ff[?] Select the firmware version and name: 1.0.1 - BK7231T / HYS_03_010_RGB_soq7i[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_1_plug_a[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_light_mi[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_ty_ffc_n[?] Select the firmware version and name: 1.0.10 - BK7231N / bk7231n_lowpower_co[?] Select the firmware version and name: 1.0.10 - BK7231N / oem_bk7231n_gate_ma[?] Select the firmware version and name: 1.0.10 - BK7231N / oem_bk7231n_light_p[?] Select the firmware version and name: 1.0.10 - BK7231T / oem_bk7231s_light_t[?] Select the firmware version and name: 1.0.14 - BK7231N / oem_bk7231n_slide_s[?] Select the firmware version and name: 1.0.15 - BK7231N / oem_bk7231n_slide_s[?] Select the firmware version and name: 1.0.18 - BK7231N / oem_bk7231n_slide_s[?] Select the firmware version and name: 1.0.2 - BK7231N / bk7231n_common_user_[?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_dctrl_sw[?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_strip_sl[?] Select the firmware version and name: 1.0.2 - BK7231T / HYS_03_010_RGB_xhsx0[?] Select the firmware version and name: 1.0.2 - BK7231T / bk7231t_common_user_[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_2_plug_g[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_dltj_tes[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_light_mi[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_rnd_zero[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_strip_ke[?] Select the firmware version and name: 1.0.3 - BK7231N / oem_bk7231n_curtain_[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_light_ty[?] Select the firmware version and name: 1.0.4 - BK7231N / oem_bk7231n_strip_li[?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_light4_4[?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_light_st[?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_ty_ffc_n[?] Select t [... cut ...]
Firmware update progress: 98%
Firmware file has been sent and MQTT reported a progress of nearly complete.  Waiting 15 seconds to ensure flashing completes.
Flashing should be complete.  It takes about 15 seconds for the device to reboot and verify the flash was valid.
Please wait about 30 seconds then look for signs of activity from the firmware you supplied (either watch for AP mode or check if it joined your network).
Device MAC address: fc:67:1f:f0:f3:7b
kurt@raspberrypi:~/tuya-cloudcutter $

Cossid, 2023-10-09:
Yes, you flashed the kickstart firmware, which is an intermediate firmware, it does not have actual device functionality.  You're now beyond the scope of cloudcutter support.  You need to make a device-specific firmware/configuration and re-flash you device with that with the tools available to the firmware you flashed.

closed 2023-10-09 by Cossid

kurtdefeyter, 2023-10-09:
update i was able to connect to the kickstart network and the device is online now   ...many thanks