tuya-cloudcutter/tuya-cloudcutter #498: YSP102 Plug never sends a firmware upgrade request

xenfasa, 2023-10-09:
This is the output using a RP4
I have tried on 2 plug switches so far with the same result.
Firmware on the plugs are 1.1.8
ideas?

`n@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -rv
Building cloudcutter docker image
[+] Building 11.1s (13/13) FINISHED                                                                                                                                            docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                     0.1s
 => => transferring dockerfile: 460B                                                                                                                                                     0.0s
 => [internal] load .dockerignore                                                                                                                                                        1.6s
 => => transferring context: 2B                                                                                                                                                          0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                                                                                             1.7s
 => [internal] load build context                                                                                                                                                        0.1s
 => => transferring context: 3.02kB                                                                                                                                                      0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6b5f1c98e869ac271786632                                                  0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                     0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                            0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                                      0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                                 0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                             0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                               0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                                0.0s
 => exporting to image                                                                                                                                                                   0.1s
 => => exporting layers                                                                                                                                                                  0.0s
 => => writing image sha256:f3615ebbae313c4d45723918fb5da73aaecde465f351432dd93bcbe58d495f93                                                                                             0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                                           0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.7 - BK7231N / CL_DREAM_RGB_STRIP_19KEY_1KEY_5V_BK7231N
   1.1.7 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.1.7 - BK7231N / oem_bk7231n_plug
   1.1.7 - BK7231T / oem_bk7231s_light_pir_ty
   1.1.7 - BK7231T / oem_bk7231s_light_ty
   1.1.71 - BK7231T / bk7231t_common_user_config_ty
 > 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.80 - BK7231T / bk7231t_common_user_config_ty
   1.1.9 - BK7231N / oem_bk7231n_water_sensor_plus
   1.1.9 - BK7231T / bk7231s_common_iot_config_ty
   1.1.9 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.1.9 - BK7231T / oem_bk7231s_strip_ir_daybetter

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231n.ug.bin

Selected Device Slug: tuya-generic-lspa7-plug
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
[... 92 lines cut ...]
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Oct  9 14:28:18 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 DHCPREQUEST(wlan0) 10.42.42.12 a8:80:55:86:21:8d 
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 tags: wlan0
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 DHCPACK(wlan0) 10.42.42.12 a8:80:55:86:21:8d wlan0
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  5
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Oct  9 14:28:20 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1

`

Cossid, 2023-10-09:
It appears to be starting the DHCP process, but not completing it.  Make sure the devices are as near each other as they can be, then trying to re-start the connect process from the FAQ: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

xenfasa, 2023-10-09:
After anther 15 minutes of turning off and on and trying different things it finally connected.   Definitely not smooth.  I guess depends on the hardware and a lot on the timing of the devices

closed 2023-10-09 by Cossid