tuya-cloudcutter/tuya-cloudcutter #499: Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device.

xenfasa, 2023-10-09:
I'm connecting another YSP102 SmartPlug.  I often get these errors below.  I'm not sure why.
Any way to prevent it?

Oct  9 16:01:50 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.11
Oct  9 16:01:50 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Oct  9 16:01:59 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.11
Oct  9 16:01:59 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Oct  9 16:02:08 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.11
Oct  9 16:02:08 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Oct  9 16:02:16 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.11
Oct  9 16:02:16 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Oct  9 16:02:22 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.11
Oct  9 16:02:22 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Oct  9 16:02:31 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.11
Oct  9 16:02:31 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Oct  9 16:02:32 dnsmasq[15]: query[A] a3.tuyaeu.com from 10.42.42.11
Oct  9 16:02:32 dnsmasq[15]: config a3.tuyaeu.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying is false - bailing out
[!] Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device.
Oh no, something went wrong with updating firmware! Try again I guess...

Cossid, 2023-10-09:
That can only happen if you cut a device with a correct profile and later use an incorrect profile, or you have a different device connecting to cloudcutter.

xenfasa, 2023-10-09:
This one was already cut but the firmware request never happened so it was never finished.... and I had to rerun cloudcutter.  What is the solution?   It seems to take a while to run cloudcutter again and again if this happens to arrive at the point to get the new firmware on it.

Cossid, 2023-10-09:
Yes, you have to go back to the profile that caused the AP SSID to change to `A-`, using anything different will always result in SSL errors and failure.

xenfasa, 2023-10-09:
I can't recover from this... I tried both profiles I potentially used and still the same error...
Is there anyway out of this?
Oct  9 16:39:08 dnsmasq[15]: query[A] a3.tuyaeu.com from 10.42.42.11
Oct  9 16:39:08 dnsmasq[15]: config a3.tuyaeu.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying is false - bailing out
[!] Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device.
Oh no, something went wrong with updating firmware! Try again I guess...

Also, is there any way have cloudcutter jump right to this part below?  
That would save a lot of time.

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Cossid, 2023-10-09:
Any error related to SSL, AUTHKEY, or SECKEY means you are using the wrong profile or an incompatible device, or you're mixing up devices.

No, there is no way to skip to that part because it sets different authkey/seckey every time you run the exploit.  And since you're apparently using the wrong profile, it's not even correctly changing it.

xenfasa, 2023-10-09:
I bought a box of 4 of these YSP102 JUNLIT smart plugs.

1) Connected to the Tuya cloud then used cloudcutter on it.  I could never get past the firmware upgrade, thought I was able to ping it  at  10.42.42.11 at one point. It is the one that now has the error.

2) After being stuck at the fireware request... then spending 15 mins on the FAQ and Tuya/SmartLife App and lots of off/ons the firmware finally loaded.  This is the only one that works now

3) Using 1.1.8 - BK7231N / oem_bk7231n_plug........
Found access point name: "SmartLife-DF54", trying to connect...
Device 'wlan0' successfully activated with 'a21bf9f2-cfeb-4778-8811-6d1c9b3e2d2e'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

4) Can't get it to request fireware still :(

Since one worked I think there is hope but not sure what to do next

xenfasa, 2023-10-09:
If the incorrect profile is chosen the first time,  is it bricked?  or can the device recover from this or any problem by starting cloudcutter over again... starting from scratch?
I'm not 100% sure how it all works.   If encryption is put on the first time... they maybe you have one shot to get it right?

Cossid, 2023-10-09:
No, you pretty much can't brick these devices without a lot of effort.

If a profile changes the device's AP SSID from `SmartLife-` to `A-`, that is the correct profile, and the only one you should be using.  If you use the wrong one, you'll just get SSL/AUTHKEY/SECKEY errors, but as soon as you use the correct one, everything will continue on as normal.  If it never changed from `SmartLife-` to `A-`, it wasn't the right profile.  This gets confusing after you successfully exploit it, but don't finish, then come back and use the incorrect profile, those SSL/AUTHKEY/SECKEY errors are wall we can show, and we have no history/context to guide things better.

xenfasa, 2023-10-09:
The real problem I am seeing is that the devices always seem to connect to the cloudcutter network and get an IP address but nothing happens after that.  It never continues.  If it has an IP than FAQ info about using the smartlife app to set the network again does not really apply... and when I do try that after it already had an IP it never wants to connect to cloudcutter AP again. 
 I never see it request another IP.      It seems it would work the first time if after connecting to the cloudcutter AP it would request the firmware or something start the upload process but that does not happen.

Cossid, 2023-10-09:
If it gets an IP but then doesn't continue on, there is nothing we can do, that is entirely either wifi signal/network condition, or tuya wifi code we cannot change.  It can't request a firmware update without going through the tuya join protocol, which we do not control, it happens automatically when it joins a wifi network, if it was successful anyway.  The only solution is to start the process over.

Cossid, 2023-10-18:
Closing as stale.  There are no changes that can be made on the cloudcutter side, this appears to be signal quality related.

closed 2023-10-18 by Cossid

AdLf0208, 2026-03-13:
Okay, I will try to use the first one which triggered the change

> > > Closing as stale. There are no changes that can be made on the cloudcutter side, this appears to be signal quality related.
> > 
> > 
> > here from 2026)) I have the same error but different case: I am able to change the AP form SmartLife to A-.... but when I continue with installation after second device's reboot I get:
> > ```
> > Found access point name: "A-9944", trying to connect...
> > Device 'wlan0' successfully activated with '466dc0af-b1e7-4e25-b71c-324bbe0400bd'.
> > Connected to access point.
> > Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
> > Configured device to connect to 'cloudcutterflash'
> > Flashing custom firmware...
> > 
> > ================================================================================
> > Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
> > ================================================================================
> > 
> > Using WLAN adapter: wlan0
> > Unable to get channel from wlan info. Using 6 as fallback.
> > Configuration file: /dev/stdin
> > Using interface wlan0 with hwaddr 94:08:53:3e:85:ad and ssid "cloudcutterflash"
> > wlan0: interface state UNINITIALIZED->ENABLED
> > wlan0: AP-ENABLED 
> > If your device gets stuck here with no progress after several (at least two) minutes, see https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do for additional steps
> > 
> > Using PSK v1 - Received PSK ID version 01
> > Processing endpoint /v2/url_config
> > Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
> > Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying is false - bailing out
> > [!] Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device - Padding is incorrect.
> > Oh no, something went wrong with updating firmware! Try again I guess...
> > If you need to report an issue, please run with the -v flag and supply the full log of that attempt.
> > ```
> 
> Decryption failing means you're no longer using the same profile that successfully changed the device from `SmartLife-` to `A-`. You must find and use the profile that was successful in cutting the device. Your log doesn't show the change happening, only that it already has an `A-` prefix.