tuya-cloudcutter/tuya-cloudcutter #500: AP Gateway

dreleith, 2023-10-11:
Message - Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got 192.168.4.1
no connection and cycle through until failed.
Is there a config issue?

Cossid, 2023-10-11:
As we already asked in discord, and you ignored and made two github issues instead...
Are you sure it is a Beken or Tuya device?
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

Tuya devices are usually not 192.168.4.1, and if they are, they are not beken devices.  Does this device already have a 3rd party firmware on it?

dreleith, 2023-10-11:
Hello, thanks for the reply and please do accept my apology, I did not see and ignore the discord response.

Confirming the BSSID lookup shows “No vendor details found.” 

 

Thanks for your support.

 

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

Cossid, 2023-10-11:
What is the bssid?  That is almost certainly an Espressif ESP chip.  If you subtract hex 2 from the first character set, it will probably match Espressif.  That means the device is not compatible with CloudCutter.

closed 2023-10-11 by Cossid

dreleith, 2023-10-11:
de:4f:22:22:d9:df
ee:fa:bc:9e:ae:43

Cossid, 2023-10-11:
dc:4f:22:22:d9:df is Espressif
ec:fa:bc:9e:ae:43 is Espressif

Both of those are ESP devices, and not compatible with Tuya-CloudCutter.  You may consider trying Tuya-Convert, and failing that, you would need to serial flash.