tuya-cloudcutter/tuya-cloudcutter #502: 2578685 970766 Smart Plug CB2S Not exploiting HELP ME PLEASE

N0edL, 2023-10-15:
im trying to cut my smart plug but it keeps failing on the second time i connect to it

Cossid, 2023-10-15:
You're not going to get any help without supplying information, at minimum the output of the script.

N0edL, 2023-10-15:
Selected Device Slug: lsc-2578685-970766-smart-plug-cb2s
Selected Profile: oem-bk7231n-plug-1.1.7-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.........
Found access point name: "SmartLife-7542", trying to connect...
Device 'wlan0' successfully activated with '98f7f1e5-8802-405f-a40f-c9fc5bb592b0'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/wtnxUJGHvQAy.deviceconfig
Saved device config in /work/configured-devices/wtnxUJGHvQAy.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
....
Found access point name: "SmartLife-7542", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Wiping NetworkManager configs
....
Found access point name: "SmartLife-7542", trying to connect...
Device 'wlan0' successfully activated with '51193c71-2d81-4df9-a237-85263083f500'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

Sorry here is everything

N0edL, 2023-10-15:
im saw a tutorial online and i followed it did what he did but i did it with a athore product but followed the same step expet the product info and now im getting this error `[!] The profile you selected did not result in a successful exploit.` so maybe you can help thanks

Cossid, 2023-10-15:
Your device appears to have a different firmware version that one that a profile version exists for.  You can start by finding what version your device actually has: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

N0edL, 2023-10-15:
Okay i see
Main Module: V1.1.8
MCU Module: V1.1.8

Cossid, 2023-10-15:
Now instead of choosing by manufacturere/device name, choose by firmware version and name, and select 1.1.8 and a plug firmware.  Probably BK7231N, but it's possible it could also be a BK7231T.  Whatever profile works, install one of the firmwares presented in order to avoid bricking your device, and continue using that chip type moving forward.

1.1.8 exists for both BK7231N and BK7231T chips.  Alternately, you could open the device and see which module your device has (not an easy step for sealed devices).

N0edL, 2023-10-15:
So if im reading this right i need to select choose by firmware version and name and search for the plug with that board an version

Cossid, 2023-10-15:
Correct, you'll see when you choose that option, the firmware version will be listed first, followed by what firmware name it is.

N0edL, 2023-10-15:
The only one i find is 1.1.8  BK7231N / oem_bk7231n_plug

Cossid, 2023-10-15:
That's the first one I'd try.  Sometimes they are referred to as `switch` though, which is where you'll find the other ones.

N0edL, 2023-10-15:
1.1.8 BK7231N / oem_bk7231n_plug NOT WORKING
1.1.8 BK7231N / oem_bk7231n_rnd_switch TRYING NOW

N0edL, 2023-10-15:
1.1.8 BK7231N / oem_bk7231n_rnd_switch 

I think its good

```

Found access point name: "A-7542", trying to connect...
Device 'wlan0' successfully activated with '7e4252df-6ae7-42e7-85e7-8d9111e16771'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr d8:3a:dd:17:a2:8e and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v1/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.uuid.pskkey.get
[MQTT Sending] Triggering firmware update message.
Processing endpoint tuya.device.upgrade.get
Processing endpoint tuya.device.upgrade.status.update
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
Firmware update progress: 30%
Firmware update progress: 60%
Firmware update progress: 90%
Processing endpoint atop.online.debug.log
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dynamic.config.ack
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin send complete, request range: bytes=0-438751/438752
Firmware update progress: 98%
Firmware file has been sent and MQTT reported a progress of nearly complete.  Waiting 15 seconds to ensure flashing completes.
Flashing should be complete.  It takes about 15 seconds for the device to reboot and verify the flash was valid.
Please wait about 30 seconds then look for signs of activity from the firmware you supplied (either watch for AP mode or check if it joined your network).
Device MAC address: cc:8c:bf:51:75:42
```

Cossid, 2023-10-15:
Yep, you got an `A-` SSID prefix, that means it's the correct profile, but it looks like it was the `BK7231T`, not `1.1.8 BK7231N / oem_bk7231n_rnd_switch`, so make sure that's the firmware you're selecting/compiling moving forward.

closed 2023-10-15 by Cossid

N0edL, 2023-10-15:
Greate but now im walking up to a new problem it wont let me install my esphome yaml with 
`output:
  - platform: libretiny_pwm
    id: output_led_1
    pin: P6`
    
in it

Cossid, 2023-10-15:
That's an issue for support of whatever custom firmware you installed, no longer a cloudcutter issue.  Not sure where you got your config from, but if it was from the BK7231N of your plug, it's likely not the same.  I'd recommend installing KickStart and LTChipTool to generate a device-specific config.  If specifically `libretiny_pwm` isn't compiling for you, you're on an old version of ESPHome.  LibreTiny was merged into the latest official ESPHome release, you shouldn't need a separate container for it anymore.