tuya-cloudcutter/tuya-cloudcutter #503: Bulb with Main & MCU v1.1.8 can not patched to OpenBeken

HalliHalloSchatz, 2023-10-17:
Hello,

I used the profile by Firmware and Version from chip.

I used bk7231n and Main & MCU v1.1.8
Version found in tuya App.

Exploit not working with that profile.
I read, that with lightleak I can try.
Profile "Standard" worked, I have a 2MB "flash.bin" from the bulb.

What are the next steps to make it possible to find the correct profile / or create the correct profile from dump_flash.bin?

Very good work and I am very happy for assistant.

Best regards, Marco

Cossid, 2023-10-17:
Attach the flash.bin here.

HalliHalloSchatz, 2023-10-17:
Which file ending should I used?
.bin are blocked?

Cossid, 2023-10-17:
either zip it up, or just add .zip to the end.

HalliHalloSchatz, 2023-10-17:
I add the binary. I will add a few information in a few minutes to the product.

Cossid, 2023-10-17:
Added as `Tuya Generic Smart Wall Light v1.1.8`, may take a few minutes to show up in the list.

closed 2023-10-17 by Cossid

HalliHalloSchatz, 2023-10-17:
Wow!

You need more information from me?
I will test it later... Is the profile automatically available or should I update / push / pull?

I used ./tuya-cloudcutter.sh

Cossid, 2023-10-17:
The script pulls the profile list from the API, but the APi takes a few minutes to refresh after a new item is added.  You don't need to pull anything.

Cossid, 2023-10-17:
Yes, you can review the data at https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/31c77b945e7790f6b30648a1e103c54a8462edba/devices/tuya-generic-smart-wall-light-v1.1.8.json#L160 or https://upk.libretiny.eu/?profile=tuya-generic-smart-wall-light-v1.1.8

HalliHalloSchatz, 2023-10-17:
I have use this here
https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/tuya-generic-smart-wall-light-v1.1.8.json
Put in the OpenBeken WebApp import, reset bulb, but I can not get light from the bulb.

Every simple and stupid try from me, all black...

Cossid, 2023-10-17:
Ah, looks like that configuration came from the firmware and not from the storage (there was not a configuration in storage).  Sometimes the ones in the firmware are the demo configurations, and don't apply.

If nothing came on while testing all modes (cold/warm white, red, green, blue), you probably do not have a PWM light, but probably an LED driver.  Unfortunately, I can't be any further help, you'll probably need to inspect the device to find out what LED driver it has, and trace/guess what pins said driver is on.

HalliHalloSchatz, 2023-10-17:
I wonder, I press the button for "storage" in the lightleak android app, but I don't get a new file in path. 

I have a "little" bigger binary file... Around 130kb... 2,13MB

Should it help?

Cossid, 2023-10-17:
Based on the firmware class of `farylink_soc_hzs_5_light_15155`, fary (or usually fairy), usually refers to RGBIC or WS2812b

Cossid, 2023-10-17:
I'm not sure if OpenBeken has RGBIC, I know ESPHome/LibreTiny is still incomplete.  Luckily, you can serial flash the original bin back, and then just cut it and use it with LocalTuya/TinyTuya as an option.

HalliHalloSchatz, 2023-10-17:
Mhh, serial cable... I think that is very very tiny chip... 
[image]

Cossid, 2023-10-17:
That is a strange module, not a standard Tuya one, so I don't know what those gold pins would map to.  Are those 8 gold pins the only ones, or are there more undder the silicon?

There is a way to do it OTA, but you'll have to dissect your dump, get the app partition, and put it into the format needed for OTA for OpenBeken, I don't fully know that process, but they might be able to help you.  Again, that is if they don't have WS2812b/RGBIC support (I have no idea there).

HalliHalloSchatz, 2023-10-17:
Yeah, first look, no more big pins under the silicon.
I will try to remove the silicon.

The little chip is the bk7231N

HalliHalloSchatz, 2023-10-17:
On the other side on that lamp, there is a little sm15155e
[image]

Cossid, 2023-10-17:
That explains 15155 in the firmware name.
http://www.chinaasic.com/chipDetails/detail_273.html
Yeah, that looks like an RGBIC driver.  Assuming it's WS2812b based on the reference in the Tuya firmware, but not certain.  If not, I haven't found a specsheet for that specific driver.

We can see there is just one data pin based on how the module is soldered.

HalliHalloSchatz, 2023-10-17:
Under the white silicone, there a no more pins...

HalliHalloSchatz, 2023-10-20:
SM15155 on BK7231N PIN16