tuya-cloudcutter/tuya-cloudcutter #504: Smart Plug not exploiting

gustavomahlow, 2023-10-18:
Hello guys. I'm trying to flash a smart plug that is sold here in Brazil.

The brand is not listed on "By manufacturer/device name" option. It's called "NovaDigital".

I've tryed several firmwares under "By firmware version and name" option, and none of them worked.

The "Smart Life" app shows me the following information.

Main Module: V1.1.15
Other type: V1.0.7

The version 1.1.15 is not listed on under "By firmware version and name".

The only thing that i know about the "inside" of the device it that is has a CB2S module.

Here is the log output for one of the firmwares that i've tryed.

`Building cloudcutter docker image
[+] Building 1.1s (13/13) FINISHED                                                                                                                                                                  docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                          0.0s
 => => transferring dockerfile: 460B                                                                                                                                                                          0.0s
 => [internal] load .dockerignore                                                                                                                                                                             0.0s
 => => transferring context: 2B                                                                                                                                                                               0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                                                                                                                  1.1s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6b5f1c98e869ac271786632                                                                       0.0s
 => [internal] load build context                                                                                                                                                                             0.0s
 => => transferring context: 3.02kB                                                                                                                                                                           0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                                          0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                                                 0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                                                           0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                                                      0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                                                  0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                                                    0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                                                     0.0s
 => exporting to image                                                                                                                                                                                        0.0s
 => => exporting layers                                                                                                                                                                                       0.0s
 => => writing image sha256:efc1cdc7ceb0be2d463d774ffcab1b67cdeff4ce6b9904c68029319650b44f38                                                                                                                  0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                                                                0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.4 - BK7231N / oem_bk7231n_plug
   1.1.2 - BK7231T / oem_bk7231s_ty_ffc_db_ldv
   1.1.2 - BK7231T / oem_bk7231t_light3_laser_nanxin
   1.1.3 - BK7231N / bk7231n-unknown
   1.1.3 - BK7231T / oem_bk7231s_double_switch_rmd
   1.1.3 - BK7231T / oem_bk7231s_irbox_mol_ty
   1.1.3 - BK7231T / oem_bk7231s_switch_curtain_rj
 > 1.1.4 - BK7231N / oem_bk7231n_plug
   1.1.4 - BK7231T / oem_bk7231s_dltj_2switch_alerc
   1.1.4 - BK7231T / oem_bk7231s_ty_ffc_db_ldv
   1.1.5 - BK7231N / oem_bk7231n_control_switch
   1.1.5 - BK7231N / oem_bk7231n_plug
   1.1.5 - BK7231N / oem_bk7231n_strip_ty
   1.1.5 - BK7231T / oem_bk7231s_4_speed_controller

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 25131.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service

Checking UDP port 67... Available.
[... 25 lines cut ...]
Found access point name: "SmartLife-1891", trying to connect...
Device 'wlp8s0' successfully activated with 'a0549aae-d34c-43f9-9c0b-c8e5c9eb1e8e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/IvFP3vf43gZN.deviceconfig
Saved device config in /work/configured-devices/IvFP3vf43gZN.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-1891", trying to connect...
Device 'wlp8s0' successfully activated with 'a0549aae-d34c-43f9-9c0b-c8e5c9eb1e8e'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
`

Cossid, 2023-10-18:
If there is no 1.1.5 profile, it will not work unless you are able to dump the firmware for the device.  You can't just choose random unmatched firmware and expect it to work.

gustavomahlow, 2023-10-18:
Sorry @Cossid I think it was a typo. The Smart Life app shows me V1.1.15, not 1.1.5.

Cossid, 2023-10-18:
I mis-read that, 1.1.**15** is on the [known patched firmware list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware).  Serial flashing will be your only option.

closed 2023-10-18 by Cossid