tuya-cloudcutter/tuya-cloudcutter #506: Firmware  V3.3.35 is missing

griperik, 2023-10-18:
Hi! I have a Nedis filament globe bulb with firmware V3.3.35. But V3.3.35 is missing. Is it possible to make support for it in Tuya Cloudcutter?
[image]

Cossid, 2023-10-18:
3.3.35 is almost certainly an ESP8266 module and not compatible with CloudCutter.  If it's not ESP8266 and is indeed Beken, it's probably going to be TuyaOS 3.x and patched.

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out can help you identify which processor yours has.

Cossid, 2023-10-25:
Closing as stale, and not likely a beken device.

closed 2023-10-25 by Cossid