tuya-cloudcutter/tuya-cloudcutter #507: `DETA 6294HA Outdoor Double Powerpoint` - Not Working

davewatson91, 2023-10-21:
Added as `DETA 6294HA Outdoor Double Powerpoint`

_Originally posted by @Cossid in https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/471#issuecomment-1707505662_

I wasn't able to get this profile working properly. I did work around it by doing a direct firmware choice (1.1.4), however the profile failed for me 10x before I gave up on it.

Cossid, 2023-10-21:
The profile that was added was for v1.1.2, it sounds like you have a different firmware version, and we would need a new firmware dump to support that device.

Cossid, 2023-10-21:
It looks like an upgrade is available from 1.1.2 to 1.1.4, I will be able to pull and re-process this, and support will be added.

Cossid, 2023-10-21:
Support for `DETA 6294HA Outdoor Double Powerpoint v1.1.4` has been added.

closed 2023-10-21 by Cossid