tuya-cloudcutter/tuya-cloudcutter #508: Deta 6922HA (MBK) Smart Double PowerPoint - New device

davewatson91, 2023-10-21:
Hi - firstly, thanks for all the great work with this! 

I've just purchased the DETA 6922HAMBK (MBK for matte black) smart double power point. I believe that this is an BK7231T (according [to ](https://[openbekeniot.github.io/webapp/devicesList.html](https://openbekeniot.github.io/webapp/devicesList.html))this page). 

I have checked on the Tuya app, and it is running firmware version 1.1.3, but unfortunately won't flash with the 3x profiles available for this. 

What are my next steps in getting this one working? Thanks!

Cossid, 2023-10-21:
If `1.1.3 - BK7231T / oem_bk7231s_double_switch_rmd` does not work, we would need a firmware dump to support this device.

davewatson91, 2023-10-21:
Thanks @Cossid, no worries - is there a guide on how I can provide a firmware dump? Can this be done OTA?

Cossid, 2023-10-21:
You can try OTA via [LightLeak](https://github.com/tuya-cloudcutter/lightleak) if you meet the requirements (android device with wifi and a dev board you can install the intermediary on), but it is a rather complex and imperfect process.  Otherwise you would need to dump via serial using ltchiptool: https://docs.libretiny.eu/docs/flashing/tools/ltchiptool/

davewatson91, 2023-10-21:
Thanks @Cossid - that was definitely a process, but I got there 😄 

I have attached the dump - let me know what else you need from me. Really appreciate the help! 

[dump_Deta_6922HA.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13062037/dump_Deta_6922HA.zip)

Cossid, 2023-10-22:
Added as `DETA 6922HA Double PowerPoint Plug v1.1.3`, it will be available in a couple minutes.

closed 2023-10-22 by Cossid

davewatson91, 2023-10-22:
That's perfect - worked immediately. Thanks!