tuya-cloudcutter/tuya-cloudcutter #509: Unable to cut Moes WS-B-US

TommySharpNZ, 2023-10-23:
I've been battling with this a bit and eventually got it to a point where it tries to update the firmware but then get this error...

```
Oct 23 21:30:50 dnsmasq[16]: query[A] a3.tuyaus.com from 10.42.42.21
Oct 23 21:30:50 dnsmasq[16]: config a3.tuyaus.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01

[21:30:50.403682 Log (Client)] Request: HTTPServerRequest(protocol='https', host='a3.tuyaus.com', method='POST', uri='/d.json?a=tuya.device.uuid.pskkey.get&et=1&t=6&uuid=aEMVRiO3C29d&v=1.0&sign=db6107e682f60d21a780ab3733031e61', version='HTTP/1.1', remote_ip='10.42.42.21')
[21:30:50.403891 LOG (Client)] ==== Request body ===
{"t":6}
[21:30:50.404666 LOG (Client)] ==== End request body ===
[21:30:50.406416 LOG (Server)] Response:  {'result': {'pskKey': ''}, 't': 1698096650, 'success': True}
[21:30:50.409805 MQTT Sending] Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1698096650} using protocol 2.2
[21:30:50.415129 MQTT Sending] Triggering firmware update message.
[21:30:50.415432 MQTT Received] Topic: smart/device/in/4wZ9YVmw4oO0Mxqabik5 - Message: {"data":{"firmwareType":0},"protocol":15,"t":1698096650}
Oct 23 21:31:10 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:31:10 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:31:10 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:31:10 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:31:30 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:31:30 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:31:33 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:31:33 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:31:59 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:31:59 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:32:04 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:32:04 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:32:26 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:32:26 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:32:32 dnsmasq[16]: query[A] m2.tuyaus.com from 10.42.42.21
Oct 23 21:32:32 dnsmasq[16]: config m2.tuyaus.com is 10.42.42.1
Oct 23 21:32:33 dnsmasq[16]: query[A] a3.tuyaus.com from 10.42.42.21
Oct 23 21:32:33 dnsmasq[16]: config a3.tuyaus.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying is false - bailing out
[!] Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device.
Oh no, something went wrong with updating firmware! Try again I guess...
```

Cossid, 2023-10-23:
`PSKKEY/AUTHKEY do not match device.` means you are using an incorrect profile.  Because the device already has an `A-` prefix, we can't protect against you using the wrong profile.

You will not get PSKKEY/AUTHKEY errors using the correct profile.  You need to go back to the profile that caused the device to change to `A-` in the first place.

TommySharpNZ, 2023-10-24:
Thanks, in amongst all my trial and error it does seem to have become mixed up. I have tried all the profiles and none work. I also have a second switch so I'll now be trying with that one to see if I can make some progress starting entirely from scratch agin.

TommySharpNZ, 2023-10-24:
Argh, the other switches a bought all seem to have the v1.3.5 tuya firmware. There is only one profile for this version and it doesn't work :-(

Cossid, 2023-10-24:
The 1.3.5 firmware for plugs is for BK7231T, if that profile didn't work, then it means you probably have BK7231N, and 1.3.5 for that one is on the known patched list.  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware  Serial flashing will probably be your only option there.

TommySharpNZ, 2023-10-24:
Thanks, they are pretty easy to open up so will investigate serial flashing.

Cossid, 2023-10-24:
Sorry, mixing up issues.  Yeah, 1.3.5 on the Moes switches is for sure patched and BK7231N.  Serial flashing is for sure the only option.

Cossid, 2023-10-27:
Closing, as your specific case requires serial flashing, there is nothing we can do on the CloudCutter side.

closed 2023-10-27 by Cossid