tuya-cloudcutter/tuya-cloudcutter #511: Cannot connect to cloudcutterflash AP on Raspberry Pi

drewkeller, 2023-10-25:
I've made several attempts on this device. It got an "A-" AP, so I guess that means it has been "cut"? This is a TreatLife SL-10 RGB bulb, which Tuya app showed as version 1.3.21 (checked before I started). Previous reports on these bulbs showed different versions (e.g. 1.0.8), which I tried first without the exploit succeeding and it seemed to work on a similar "Globe" bulb with that version number.

At the cloudcutterflash portion of the process, the bulb doesn't complete the connection to the cloudcutterflash AP. I tried with my phone also and it says **abcdabcd** is the wrong password.

Verbose terminal output is [here](https://pastebin.com/Wi398NrR).

I attempted to connect to the bulb using the Tuya app ("add" function), but that didn't find the bulb, either. I'm assuming it should still recognize the device's Bluetooth pairing?

I successfully converted/programmed several tasmota devices on this Raspberry Pi recently (tuya-convert).

Cossid, 2023-10-25:
abcdabcd is the correct password ( see https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/4aa91be237b811e24c71ab15eb8bb6b009d093b6/src/setup_apmode.sh#L42 for where we set it ).  Make sure you are in range (AP mode is very weak) and your phone keyboard is not changing case or trying to autocorrect.

Tuya app should see the device in bluetooth pairing mode, but will not actually add it (but it should attempt to make it connect to cloudcutterflash), but I still highly recommend using the slow-blink method instead of the bluetooth method.

Are you already following the directions at https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do ?

drewkeller, 2023-10-26:
Thanks for the quick response!

My RaspberryPi and the bulb are both on my desk. Yes, I've read through the troubleshooting page and attempted the section you mentioned.

I got a little farther today. The bulb got to DHCPACK stage and IP address assigned, but the firmware still wasn't getting downloaded. I thought the difference might be because I was using a wifi dongle (wlan1) on my pi but then I also got to DHCPACK using onboard wifi (wlan0) like before, so maybe not.

My phone was able to connect to the AP when it was running on wifi dongle, but not onboard wifi. Both of the Tuya apps I have on my phone told me the cloudcutterflash network was not usable for adding a device when using either quick flash or slow flash (I presume because of lack of internet?), and the app would not let me continue past that point.

Another thing I tried was using a second Raspberry Pi to connect to the cloudcutter pi instead of my phone. It was constantly connected/disconnected when cloudcutter pi was using wlan0, but did get a stable connection when it was using wlan1. The Tuya app runs on Android unfortunately, so hmmm.

I also tried an older phone of mine, but it wouldn't connect to the AP and is too old for the apps, anyway.

Here is a bit of progress on the connection...
```
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 DHCPACK(wlan0) 10.42.42.19 a8:80:55:e9:06:89 wlan0
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 next server: 10.42.42.1
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  1 option: 53 message-type  5
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option: 54 server-identifier  10.42.42.1
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option: 51 lease-time  12h
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option: 58 T1  6h
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option: 59 T2  10h30m
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option:  1 netmask  255.255.255.0
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option: 28 broadcast  10.42.42.255
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option:  3 router  10.42.42.1
Oct 26 03:08:20 dnsmasq-dhcp[15]: 1045855041 sent size:  4 option:  6 dns-server  10.42.42.1
```

All kinds of output similar to this from the second Raspberry Pi:
```
Oct 26 03:52:10 dnsmasq[15]: query[A] www.googleapis.com from 10.42.42.27
Oct 26 03:52:10 dnsmasq[15]: config www.googleapis.com is 10.42.42.1
SSL Error on 16 ('10.42.42.27', 33992): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.27', 33994): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)

```

Cossid, 2023-10-26:
Not much I can advise, these types of issues are outside the control of CloudCutter.

The only two suggestions I have are
1) Try alternate AP channels, by default, it is on channel 1 in case it is an interference issue.  This can be adjusted in src/setup_apmode.sh
2) Specifically try the 2022-04-04 Bullseye build of RaspbianOS.

Cossid, 2023-10-27:
> Question I had at this point was, is there a way to send the firmware outside of the main script?

Nope, it specifically needs to go through a progression of actions, it's not a single step to firmware upgrade.

> Then I downloaded the old [2022-04-04 Bullseye](https://downloads.raspberrypi.com/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2022-04-07/2022-04-04-raspios-buster-armhf.img.xz) you suggested and set it up fresh per HOST_SPECIFIC_INSTRUCTIONS.md and had success on the first try.

Excellent!  I've remained on that version mostly just out of lazyness, but I might have to start a new install to see if I can actually finally reproduce the stall issue, in which case I might finally be able to directly test it rather than through proxy.  The thing that baffles me though is this is in a container, so I wonder if it's an issue with docker somehow, as the versions of things running within should be consistent.

> Now I am trying to figure out how to get the device into Home Assistant / ESPHome, which seems to be insisting I need to connect the device using a cable.

No need for a cable.  If you have kickstart installed on it, you can compile your firmware and just upload it to the kickstart web interface.  After that if your firmware was compiled correctly, it will be discovered by HA automatically.

Can you please verify which version of Raspbian you had that was causing the problem?

Since switch RaspbianOS version resolved your issue, I'm closing this ticket, as that is the ultimate solution for the time being.

closed 2023-10-27 by Cossid

drewkeller, 2023-10-27:
It was a pretty fresh instance, BUT it had tuya-convert and a "reverse" wifi password cracker I wrote to figure out what password I incorrectly entered into a tuya device (turns out I had entered it correctly after all, and had just forgotten that I enabled MAC filtering on my router). The latter used hostapd so maybe I left something badly configured, although the changes were really just to write different passwords into it and see if the device connected.

(32-bit)
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ uname -a
Linux 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linu                                               x
```

drewkeller, 2023-10-28:
Many thanks to you @Cossid , I was able to flash both of my RGB lights and got them working with Home Assistant.

The final hurdles were leaped by following the steps in [this video](https://www.youtube.com/watch?v=sSj8f-HCHQ0).

I'm not sure that the BK7231N variant firmware is the right one for these. The FCC number ends with WB8P, which according to [this](https://templates.blakadder.com/treatlife_SL10.html), indicates they have the WB8P chip in them, which according to various sources (#184, [here](https://www.elektroda.com/rtvforum/topic3951016.html), [here](https://www.elektroda.com/rtvforum/topic4007596.html)), means the BK7231T variant firmware should be used.

I used the kickstart-bk7231n firmware with pinscan and confirmed that the yaml I got from ltchiptool had the right pin setup (The profile for the [Globe 35798 A19 RGBCT v13.21](https://upk.libretiny.eu/) was not).

I don't know all the ins and outs, but it looks like the pins on T and N are the same but memory layout may be different, according to [unofficial BK7231 datasheet, pinout, programming, specification, wiki (BK7231T, BK7231N)](https://www.elektroda.com/rtvforum/topic3951016.html). Beken doesn't seem to have datasheets for these chips on their website.

Cossid, 2023-10-28:
It is not uncommon for the FCC information to be outdated, as these cheap manufacturers change the modules unfortunately too frequently without updating documentation.

Follow what CloudCutter and Kickstart are instructing, as it is absolutely always 100% correct.  It tracks the chip based on the exploit used, and you can't exploit a BK7231T device with a BK7231N profile or vice versa.  If you still have Kickstart installed or put it back on, you can use ltchiptool to export the storage data under the upk2esphome tab (and post it here, we can possibly add direct device support) and if the device has a upk section, it will tell you what chip is actually in the device in the device under the "module" property.

Luckily, CloudCutter and ESPHome will not let you flash firmware for the wrong chip unless you try really really hard.

drewkeller, 2023-10-28:
If I get some more of these lights, I will post the data out of ltchiptool. The family seemed to be enjoying the one I had working yesterday, so I may get some more.

Cossid, 2023-10-28:
You actually don't need to reinstall kickstart, you can add the following:

```yaml
external_components:
  source: github://libretiny-eu/esphome-kickstart
  components:
    - hub_api
  refresh: 10 min

hub_api:
```

to your config, and you can use ltchiptool just the same as if you had kickstart installed.  (then you can remove that block and re-upload when done)

drewkeller, 2023-10-28:
Is this the right data? This shows CB8P.

ltchiptool shows this warning, but only 4 pins actually do anything per my testing with pinscan one at a time. Also, ESPHome has a page for rgbw, rgbww, rgbct but not rgbcw, so I don't know what that means, or if it can be matched to something better than rgbw.
`W: UPK: Module 'cmod': rgbcw doesn't match platform rgbw`

```
{
    "gw_bi": {
        "uuid": "p5a384pwXsUq",
        "psk_key": null,
        "auth_key": "mdHFG6j2givJ92za",
        "ap_ssid": "A",
        "ap_passwd": null,
        "country_code": null,
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false,
        "fac_pin": null
    },
    "user_param_key": {
        "Jsonver": "1.0.0",
        "b_lv": 1,
        "b_pin": 9,
        "brightmax": 100,
        "brightmin": 5,
        "c_lv": 1,
        "c_pin": 24,
        "cagt": 5,
        "category": "0505",
        "cmod": "rgbcw",
        "colormax": 100,
        "colormin": 10,
        "crc": 51,
        "cwtype": 0,
        "defbright": 100,
        "defcolor": "c",
        "deftemp": 0,
        "dmod": 0,
        "g_lv": 1,
        "g_pin": 8,
        "gmkb": 60,
        "gmkg": 60,
        "gmkr": 80,
        "gmwb": 75,
        "gmwg": 70,
        "gmwr": 100,
        "mixway": 3,
        "module": "CB8P",
        "notdisturb": 0,
        "onoffmode": 1,
        "pmemory": 1,
        "prodagain": 1,
        "pwmhz": 2000,
        "r_lv": 1,
        "r_pin": 7,
        "remdmode": 0,
        "rgbt": 5,
        "rstbr": 50,
        "rstcor": "c",
        "rstnum": 3,
        "rsttemp": 0,
        "tempmix": [
            [
                "T18",
                "000",
                "000",
                "000",
                "3e8"
            ],
            [
[... 388 lines cut ...]
        },
        {
            "type": "obj",
            "mode": "wr",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 28
        },
        {
            "mode": "rw",
            "id": 31,
            "type": "raw"
        },
        {
            "mode": "rw",
            "id": 32,
            "type": "raw"
        }
    ],
    "em_sys_env": "bk7231n",
    "mf_test_close": true
}
```

Cossid, 2023-10-28:
Yep, that is correct, and confirms they swapped the WB8P with the pin compatible newer CB8P, which is indeed BK7231N

I'll add a new cloudcutter profile shortly.

Cossid, 2023-10-28:
> ltchiptool shows this warning, but only 4 pins actually do anything per my testing with pinscan one at a time. Also, ESPHome has a page for rgbw, rgbww, rgbct but not rgbcw, so I don't know what [...]

Yeah, this is a non-standard bulb.  It is actually a `rgbw` bulb, and one of your white channels is emulated by adding in color LEDs in white mode.  Unfortunately, this can be a little tricky to do with ESPHome.  That is the the "tempmix" property was doing, it was replacing a white channel with defined color channels and a percentage of the white channel.