tuya-cloudcutter/tuya-cloudcutter #514: Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got 192.168.175.1

CSSMWX, 2023-10-28:
Hi there,
SWP86-01OG
 https://www.elektroda.pl/rtvforum/topic3994030.html
 tuya-generic-swp86-01og-energy-monitoring-plug.json

 getting this error 
`Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got 192.168.175.1`
Does that error make any sense?

I can see "SmartLife-528C" on my laptop but it doesn't issue an IP.
However, this small socket's AP's speed is over 412GB!!!

[image]
Maybe my old laptop is too slow for that?

Tried/searched many things, script with SSID, without, etc pp, am at ends, what am I doing wrong?

`
admin@raspberrypi:/etc/NetworkManager/tuya-cloudcutter $ sudo ./tuya-cloudcutter                                                                                                    .sh -s tasmo 12345678 -p tuya-generic-swp86-01og-energy-monitoring-plug
Building cloudcutter docker image
[+] Building 4.4s (13/13) FINISHED                               docker:default
 => [internal] load .dockerignore                                          0.6s
 => => transferring context: 2B                                            0.0s
 => [internal] load build definition from Dockerfile                       0.4s
 => => transferring dockerfile: 460B                                       0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-bus  1.5s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:83  0.0s
 => [internal] load build context                                          0.3s
 => => transferring context: 3.02kB                                        0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                   0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipe  0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                 0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                  0.0s
 => exporting to image                                                     0.2s
 => => exporting layers                                                    0.0s
 => => writing image sha256:7191977b788dc1ba80b3a9da77d61e699dd0b1d7d66ca  0.2s
 => => naming to docker.io/library/cloudcutter                             0.1s
Successfully built docker image
Loading options, please wait...

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 699.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: tuya-generic-swp86-01og-energy-monitoring-plug
Selected Profile: ua-app-1.0.10-sdk-2.3.3-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by                                                                                                     either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repea                                                                                                    t 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, t                                                                                                    hen releasing, and then holding the power/reset button again until the device st                                                                                                    arts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
...........
Found access point name: "SmartLife-528C", trying to connect...
Device 'wlan0' successfully activated with '5f300731-849a-4d8d-a9fb-e913d0456795'.
Connected to access point.
[... 272 lines cut ...]
Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got

192.168.175.1
.
Found access point name: "SmartLife-528C", trying to connect...

(process:5906): GLib-GIO-WARNING **: 01:05:30.743: ../../../gio/gdbusobjectmanagerclient.c:1589: Processing InterfaceRemoved signal for path /org/freedesktop/NetworkManager/AccessPoint/605 but no object proxy exists

(process:5906): GLib-GIO-WARNING **: 01:05:30.746: ../../../gio/gdbusobjectmanagerclient.c:1589: Processing InterfaceRemoved signal for path /org/freedesktop/NetworkManager/AccessPoint/606 but no object proxy exists
Device 'wlan0' successfully activated with '69fbe7f2-a9e7-4489-9a8b-169c801a46cc'.
Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got

192.168.175.1
.
Found access point name: "SmartLife-528C", trying to connect...

(process:6117): GLib-GIO-WARNING **: 01:05:44.115: ../../../gio/gdbusobjectmanagerclient.c:1589: Processing InterfaceRemoved signal for path /org/freedesktop/NetworkManager/AccessPoint/628 but no object proxy exists
Device 'wlan0' successfully activated with '32aac968-784c-4dea-a0d1-ade56f5e08e6'.
Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got

192.168.175.1
Error, could not connect to SSID.
Failed to connect, please run this script again
admin@raspberrypi:/etc/NetworkManager/tuya-cloudcutter $
`

Cossid, 2023-10-28:
What OS is this raspberrypi running (and does it have a language pack installed)?  It appears to be adding newline characters in the response to looking up the gateway IP, and this is at least the second case of this.  Those glib warnings are also concerning.

Either way, the exploit was unsuccessful, as the AP name of the ssid did not change, and that's the next message it would have given.

CSSMWX, 2023-10-28:
Yes, you got it, it adds a newline each time between 
192.168.175.1 or 192.168.176.1 but got
and 
192.168.175.1

`PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian`

The -r option you kindly suggested changed the output, but still not working

used this:
`sudo ./tuya-cloudcutter.sh -r -p tuya-generic-swp86-01og-energy-monitoring-plug -s tasmo 12345678`

`admin@raspberrypi:/etc/NetworkManager/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -r -p tuya-generic-swp86-01og-energy-monitoring-plug -s tasmo 12345678
Building cloudcutter docker image
[+] Building 5.2s (13/13) FINISHED                                                                                                                                   docker:default
 => [internal] load .dockerignore                                                                                                                                              0.2s
 => => transferring context: 2B                                                                                                                                                0.1s
 => [internal] load build definition from Dockerfile                                                                                                                           0.5s
 => => transferring dockerfile: 460B                                                                                                                                           0.1s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-buster                                                                                                   3.0s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:830e161433edfe047a23ebc99c12ee0eb1dc0a50e6b5f1c98e869ac271786632                                        0.0s
 => [internal] load build context                                                                                                                                              0.3s
 => => transferring context: 3.02kB                                                                                                                                            0.1s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto           0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                  0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                            0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                       0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                   0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                     0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                      0.0s
 => exporting to image                                                                                                                                                         0.2s
 => => exporting layers                                                                                                                                                        0.0s
 => => writing image sha256:7191977b788dc1ba80b3a9da77d61e699dd0b1d7d66caf2edefc676427d18b3d                                                                                   0.1s
 => => naming to docker.io/library/cloudcutter                                                                                                                                 0.1s
Successfully built docker image
Loading options, please wait...

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by dnsmasq with PID 515.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 687.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: tuya-generic-swp86-01og-energy-monitoring-plug
Selected Profile: ua-app-1.0.10-sdk-2.3.3-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
[... cut ...]
Scanning for open Tuya SmartLife AP
...
Found access point name: "SmartLife-528C", trying to connect...
Device 'wlan0' successfully activated with 'acbd7b52-9841-4db0-aa42-9e1504ece824'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
admin@raspberrypi:/etc/NetworkManager/tuya-cloudcutter $
`

Cossid, 2023-10-28:
Well, it's working correctly now, but as I said before, it's just the wrong profile.  This device may not be supported yet.
Do you know which firmware version your device has?  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

CSSMWX, 2023-10-28:
Thanks for the -r, here is the profile file I got:
[tuya-generic-swp86-01og-energy-monitoring-plug.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13196549/tuya-generic-swp86-01og-energy-monitoring-plug.json)

Is there a way to find out firmware version of this device without downloading an app?
Or is there an open source app which I could load to check FW version?
I'm reluctant to install a chinese app on my phone.

Cossid, 2023-10-28:
No, there is no other known way to check the firmware version without dumping the firmware itself.

CSSMWX, 2023-10-28:
I'd rather bump FW than install the app.
I have access to the pins and serial usb converters etc, how can I dump FW please?

I got the json file here:
`https://openbekeniot.github.io/webapp/devicesList.html`
select 
Type: cloudcutter profile 
and 
Keyword filter: swp86

Cossid, 2023-10-28:
https://docs.libretiny.eu/docs/flashing/dumping/

CSSMWX, 2023-10-28:
Bummer, got warning already while downloading ltchiptool-v4.6.0.exe , had to override. 
Checked the file and this:
[image]

https://github.com/libretiny-eu/ltchiptool/releases
used v4.6.0 (Latest)
https://www.virustotal.com/gui/home/upload

Thanks for your help, but I give up, just too much hassles with Tuya.
People were right, these Tuya chips are not worth it, returning all units.
No more Tuya chips for me.

Cossid, 2023-10-28:
The virus warning is because of the python to exe wrapper, but that is a clean open source app you can verify and compile yourself.  It is certainly your choice though.

closed 2023-10-28 by Cossid

CSSMWX, 2023-11-03:
Hi again, meanwhile ltchiptool doesn't seem to alert/contain viruses.
[https://github.com/libretiny-eu/ltchiptool/issues/10#issuecomment-1792339985](url)
I sat down to connect my USB-Serial adapter but found that CEN is hardwired to VCC! 
No reset possibility
And RXD1 is used, too.

It seems that tuya/vendors don't want us to free their units.
Here is the pinout of the Xenon SM-PW801-U2 
[https://www.amazon.com/dp/B08Q7YM4MP](url)

8pin dual row 19mm header
                         VCC  VCC
                         GND  GND
                         SW1  SW1-R24-Q5b,Q5e-GND,Q5c-LED
 Q6c-LED,Q6e-GND,Q6b-R25-SW2  SW2

WB3S:
      VCC-CEN   1                     16  TXD1
          ADC3  2                     15  RXD1-U4_Pin5  
      VCC-EN    3                     14  PWM2
          P14   4                     13  PWM3-SW2                         
          PWM5  5                     12  RXD2-R32-VCC
  U4_Pin6-PWM4  6                     11  TXD2 
      SW1-PWM0  7                     10  PWM1-R31-GND
      VCC-VCC   8                      9  GND-GND
                             CS
                 1   2   3   4   5   6   

Q5,6  1AM          NPN General Purpose Transistor MMBT3904L
MMBT3904LT1-D
[Uploading MMBT3904LT1-D.PDF…]()

U4    BS83A04A-3   Touch I/O 8MHz Flash MCU, 63 instructions
BS83A02A-4-HoltekSemiconductor
[Uploading BS83A02A-4-HoltekSemiconductor.pdf…]()

  U4_Pin6-PWM4
  U4_Pin5-RXD1
  U4_Pin1 FingerTouch Key1
  U4_Pin2 FingerTouch Key2

R21,22         2K
R24,25,31,32  10K
R27           5K6
R29,28         1K

CSSMWX, 2023-11-03:
bummer, no Courier font?
[image]