tuya-cloudcutter/tuya-cloudcutter #515: Generic RGBCW Ceiling Light

banter7702, 2023-10-30:
Hi,

I flashed a generic rgbww ceiling light, identical with this one:
[[DUMP] [CB3S] [BK7231N] Eacam 24W RGBCW Ceiling ](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/361)

So, I had this same initial problem in that topic where pins cant be found or extracted, I used esphome kickstarter and ltchiptool to extract it but same problem.

So I manually search it and found all the rgb cw ww pwms.

But it has problems. The rgb is not optimal. Sometimes it will turn off if I fully go to 2700k or 6500k, I had to go to red to make color temp work properly. Same goes to fully going down to lowest brightness, it turns off.

And my biggest gripe is that, I lost the remote functionality. We'll, if it would be me, its fine, but the people in the house needs it. How do I restore it? And some of its fancy features such as scenes?

Here is my esphome config:

```
esphome:
  name: ceiling-lamp
  friendly_name: Ceiling Lamp

bk72xx:
  board: generic-bk7231n-qfn32-tuya

# Enable logging
logger:

preferences:
  flash_write_interval: 10s

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Ceiling-Lamp Fallback Hotspot"
    password: "redacted"

output:
  - platform: ledc
    id: output_red
    pin: P8
  - platform: ledc
    id: output_green
    pin: P24
  - platform: ledc
    id: output_blue
    pin: P26
  - platform: ledc
    id: output_cold
    pin: P7
  - platform: ledc
    id: output_warm
    pin: P6

light:
  - platform: rgbww
    id: light_rgbww
    name: Light
    color_interlock: true
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm
    restore_mode: RESTORE_DEFAULT_OFF
``` 

Here are some pics for reference:
[image]
[image]

Cossid, 2023-10-30:
This is far beyond the scope of Tuya-CloudCutter, as it has been run and completed.

It sounds to me like you either A) have the channels mixed up, or B) have a rgbct pwm layout instead of rgbww, or quite possibly both.

As for the remote, I highly doubt that is supported by any of the available 3rd party firmwares, but if it is, you'd have to use their support systems.  It sounds like you went straight to 3rd party firmware instead of just cutting, which is what is generally recommended with devices with unusual/unsupported features, but once you're on 3rd party firmware, there is no way back.

closed 2023-10-30 by Cossid

banter7702, 2023-10-30:
Hi, Cossid.

Thank you for clarifying everything. Please bear with me, Im very newbie to this. Im just following some guide online on how to de-tuya my devices.

I never heard we could just simply cut tuya cloud (yeah the tuya-cloudcutter name is shouting) instead of going straight to 3rd party firmware.

My main problem with tuya with local tuya in home assistant is they keep on going unavailable.

Will just cutting cloud solves this problem? If so, I might just go with that route so I wont have to complicate things and loose features.

Thank you for analyzing P26, I'll try to experiment with remote-receiver.

Cossid, 2023-10-30:
Once you've flashed 3rd party firmware, cutting is no longer an option.  Cutting isn't perfect either, as it relies on Tuya's firmware for connecting to wifi and such, but it does sometimes suffer from delays or lag when trying to access the services that have been cut.  If you were having issues with LocalTuya (vs the Tuya Cloud integration), you'd probably continue to have issues as well.

I was wrong on P26, that is one of  your PWM pins.  Looks like maybe P14 for the remote_receiver.

banter7702, 2023-10-30:
> Once you've flashed 3rd party firmware, cutting is no longer an option. Cutting isn't perfect either, as it relies on Tuya's firmware for connecting to wifi and such, but it does sometimes suffer from delays or lag when trying to access the services that have been cut. If you were having issues with LocalTuya (vs the Tuya Cloud integration), you'd probably continue to have issues as well.
> 
> I was wrong on P26, that is one of your PWM pins. Looks like maybe P14 for the remote_receiver. I'm not even sure it's hooked up to GPIO, so maybe it is something else entirely.

Ok, If I understood correctly, cutting means no going back to stock as well?

Thank you for clarifying the issues with cloudcutting vs local tuya.

Ok, I'll try the P14. Thank you so much!

Cossid, 2023-10-30:
Cutting means never hooking up to Tuya cloud services again, but still using the original firmware that came on your device, with original functionality.

Flashing 3rd party firmware is a step further, you can't get the original functionality, you have to replicate it with features available in the firmware you chose (if they exist).  You cannot go back to the original firmware once you have flashed with 3rd party firmware.

banter7702, 2023-10-30:
Ok, sorry if I dont understand it very much. But that means, I still could not go back to connecting to tuya's cloud (just IN CASE the need of it) if I use cloudcutter? The answer seems obvious since it is stated in the doc:
```
Using cloudcutter means that you will NO LONGER be able to use Tuya's apps and servers. Be absolutely sure that you are never going to use them again!
``` 
Just want to confirm.

The only problem I really have is the intermittent connection using the stock firmware even if it is connected to cloud. My same device flashed with 3rd party seems very stable, no request timeout. However, I'm too overwhelmed on how to optimize the functionality as stock as possible especially the remote.

And by the way, I just tried P14, inverted, non inverted, no luck. No raw messages. Do you have more suggestion?

Thank you very much for answering my queries.

Cossid, 2023-10-30:
Correct, there is no undoing cutting, it is permanent.

And based on the solder pads, I don't think the remote is on GPIO at all and I have no other suggestions.

banter7702, 2023-10-31:
Hi, I have used cloudcutter for my spare rgbww light with just cloud cutting. Why haven't I done this before?

Though the connection seems to have improved, I noticed some of weird issues that might be just related to this specific light.

I'm still using local tuya for control. It has this issue where datapoints could not be found unless you "wake them up" using the remote, pressing the off on, rgb, w, ww, scenes, so that these datapoints will show up.

But here is more frustrating, they will sleep again after some time, and the entity becomes unavailable, I have to "wake" these datapoints for them to be controllable again in HA/Local Tuya.

Im not saying the issue is related to cloudcutting. Im pretty sure it has been the issue all along for my connection issue.

If I allow outbound, this issue seems non existent. I can see that after cloud cutting, the device still phones home with port 8886 and 443. Do you think it is fine? Im not comfortable this device phoning home without knowing what datas are they sending/receiving. I heard some has been used for botnet before, and I dont like to be a part of that.

Cossid, 2023-10-31:
8886 is mqtt (secure)
443 is their psk API

It's up to you if you block them or not, but everything tuya sends is encrypted with keys, and Tuya only knows the keys that were originally installed on the device.  CloudCutter replaces all of those keys and uses the ones we assign (the same ones you use to communicate with LocalTuya), and to Tuya it will just look like gibberish because they don't know the keys, but I'm guessing their firmware has different cases for timeouts versus a 400 error you would likely get from the gibberish requests.

I can't speak to the "waking up" for datapoints.  That sounds more like a power saving mode or something, or a limitation in LocalTuya.  I'd suggest opening an issue with details on their project.

Cossid, 2023-10-31:
It's possible we can override the url/ip they use for 8886 and 443, I'll look into that when I have some free time.  If they are something we can adjust, you would be able to re-cut with the new code to take effect.  Can't promise that will be the case though.

banter7702, 2023-10-31:
Wow, Im learning a lot. Thank you for sharing these.

I think it will be nice to have these 8886/443 url/ips to have an option to change them "JUST IN CASE", but it would be easier just to block them in the firewall, no?

But in the case with my observation where these ports have an outbound access and datapoints do not sleep, do you think the option to change the 8886/443 urls is a nice to have, so we could "possibly" intercept the response needed? Maybe they just need some OK response? And to be honest, Im not still sure if this is the case. Maybe its just coincident, because right now, I blocked them again, and seems to respond fine longer than expected.

Im not sure what Im talking about. Please bear with me. Thank you very much.

You're right, this might be a power saving thing and I must bring this issue to Local Tuya as well.

Cossid, 2023-11-03:
Well, I got some time to try it out, and while we can change the values for 8886/443, it breaks the ability to flash 3rd party firmware (and I can't explain why).  So it is going to have to be left as-is for now.  So to recap, while it technically is attempting to connect to Tuya over the internet, it will always be sending gibberish and never actually be able to validate/connect to their services.

banter7702, 2023-11-04:
Understood! 

It is preferable to be left as is if it breaks things. You're right, having it open is not a big deal as they send gibberish. And there is firewall if really that paranoid.

Thank you very much. More power to you and this project! You are amazing!