tuya-cloudcutter/tuya-cloudcutter #520: 4ch rf

nookbg, 2023-11-08:
[Amazon Info](https://www.amazon.co.uk/Switch-Control-Wireless-Compatible-Googles-White/dp/B09X4HSBJS)

{
    "gw_bi": {
        "uuid": "b8b7af3051e0618f",
        "psk_key": "wVcoNCrsZnh6JbRaP0kOS5De1Ze3FRIc6pw5n",
        "auth_key": "sU11if6gX8N4PG8BA801flspX5VTgTjV",
        "ap_ssid": "SmartLife",
        "ap_passwd": null,
        "country_code": "CN",
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false
    },
    "user_param_key": {
        "backlit_dp": 16,
        "backlit_io_lv": 0,
        "backlit_io_pin": 6,
        "backlit_select": 1,
        "bt1_lv": 0,
        "bt1_pin": 26,
        "bt2_lv": 0,
        "bt2_pin": 24,
        "bt3_lv": 0,
        "bt3_pin": 11,
        "bt4_lv": 0,
        "bt4_pin": 10,
        "bt_type": 0,
        "cd_flag2": 1,
        "ch_cddpid1": 7,
        "ch_cddpid2": 8,
        "ch_cddpid3": 9,
        "ch_cddpid4": 10,
        "ch_dpid1": 1,
        "ch_dpid2": 2,
        "ch_dpid3": 3,
        "ch_dpid4": 4,
        "ch_flag1": 1,
        "ch_flag3": 1,
        "ch_flag4": 1,
        "ch_num": 4,
        "crc": 67,
        "cyc_dpid": 17,
        "dctrl_select": 0,
        "inch_dp": 19,
        "init_conf": 14,
        "jv": "1.3.3",
        "led_dp": 15,
        "module": "WB3S",
        "mxcl_led_m": 0,
        "net_trig": 1,
        "netled_lv": 0,
        "netled_pin": 1,
        "netn_led": 0,
        "nety_led": 0,
        "rand_dpid": 18,
        "reset_t": 7,
        "reuse_led_m": 0,
        "rl1_lv": 1,
        "rl1_pin": 9,
        "rl2_lv": 1,
        "rl2_pin": 14,
        "rl3_lv": 1,
        "rl3_pin": 8,
        "rl4_lv": 1,
        "rl4_pin": 7,
        "total_stat": 0
    },
    "gw_wsm": {
        "nc_tp": 9,
        "ssid": "QkcgUE9XRVI=",
        "passwd": "QW5nZWwyMDA2",
        "md": 0,
        "random": 0,
        "wfb64": 1,
        "stat": 2,
        "token": "bAUuQiL7",
        "region": "EU",
        "reg_key": "2sfZ",
        "dns_prio": 2
    },
    "is_stride": 0,
    "gw_di": {
        "abi": 0,
        "id": "bf5a630938fea8d526agaa",
        "swv": "1.0.7",
        "bv": "40.00",
        "pv": "2.2",
        "lpv": "3.3",
        "pk": "key7axydcvmea3x9",
        "firmk": "key7axydcvmea3x9",
        "cadv": "1.0.2",
        "cdv": "1.0.0",
        "dev_swv": "1.0.7",
        "s_id": "000002gki4",
        "dtp": 0,
        "sync": 0,
        "attr_num": 0,
        "mst_tp_0": 0,
        "mst_ver_0": null,
        "mst_tp_1": 0,
        "mst_ver_1": null,
        "mst_tp_2": 0,
        "mst_ver_2": null,
        "mst_tp_3": 0,
        "mst_ver_3": null,
        "dminfo_name": null,
        "dminfo_code": null,
        "dminfo_report_code": null,
        "dminfo_sn": null
    },
    "timer_arr": {
        "lastFetchTime": 0,
        "cnt": 0
    },
    "tls_ca_cnt": 0,
    "gw_ai": {
        "key": "8OyP[0>qYoDGQt[c",
        "lckey": "zF!'B]>=7qm_db7A",
        "h_url": "http://a.tuyaeu.com/d.json",
        "h_ip": "18.184.209.125",
        "hs_url": null,
        "hs_ip": null,
        "hs_psk": "https://a3.tuyaeu.com/d.json",
        "hs_psk_ip": "18.185.182.159",
        "mqs_url": null,
        "mqs_ip": null,
        "mq_url": "m2.tuyaeu.com:1883",
        "mq_ip": "52.57.38.165",
        "ai_sp": null,
        "mq_psk": "m2.tuyaeu.com:8886",
        "mq_psk_ip": "52.57.38.165",
        "time_z": "+00:00",
        "s_time_z": "[[1711846800,1729990800],[1743296400,1761440400]]",
        "wx_app_id": null,
        "wx_uuid": null,
        "dy_tls_m": 0,
        "cloud_cap": 1025
    },
    "save_off_stat": {
        "power": [
            false,
            false,
            false,
            true
        ]
    },
    "000002gki4": [
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 1
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 2
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 3
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 4
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 7
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 8
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 9
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 10
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "range": [
                    "off",
                    "on",
                    "memory"
                ],
                "type": "enum"
            },
            "id": 14
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "range": [
                    "relay",
                    "pos",
                    "none"
                ],
                "type": "enum"
            },
            "id": 15
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 16
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 17
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 18
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 19
        }
    ],
    "astro_timer": {
        "timestamp": 1698980746,
        "index": 0
    },
    "em_sys_env": "BK7231S_2M",
    "mf_test_close": true,
    "backlit_save": {
        "def_backlit_en": 1
    }
}

Cossid, 2023-11-08:
Again, we need the full dump, there are multiple firmware version 1.0.7 for plugs, and we cannot tell which is needed from storage data alone.

nookbg, 2023-11-08:
[Uploading tuya 4channel-rf.zip…]()

Cossid, 2023-11-08:
Added as `Tuya Generic TYWB-4CH-RF 4 Channel Relay`, please allow a few minutes for the API device list to update.

closed 2023-11-08 by Cossid