tuya-cloudcutter/tuya-cloudcutter #521: k601

nookbg, 2023-11-08:
{
    "gw_bi": {
        "uuid": "a1f377d4f01f7de2",
        "psk_key": "7ES8iCWz0C6isXlKbS5RXZbVtmiAP6JZ5bjLh",
        "auth_key": "eNFQK1HqW9SEurGt9lufC9yPkZFtZCAz",
        "ap_ssid": "SmartLife",
        "ap_passwd": null,
        "country_code": "CN",
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false,
        "fac_pin": "smfx3moljara13w0"
    },
    "user_param_key": {
        "ch_cddpid1": 9,
        "ch_dpid1": 1,
        "ch_num": 1,
        "crc": 84,
        "cyc_dpid": 43,
        "inch_dp": 44,
        "init_conf": 38,
        "jv": 101,
        "module": "CB2S",
        "net_trig": 4,
        "net_type": 0,
        "netled1_lv": 1,
        "netled1_pin": 26,
        "netn_led": 2,
        "nety_led": 1,
        "on_off_cnt": 10,
        "onoff1": 10,
        "onoff_clear_t": 10,
        "onoff_rst_m": 0,
        "onoff_rst_type": 2,
        "onoff_type": 0,
        "rand_dpid": 42,
        "remote_add_dp": 49,
        "remote_io": 24,
        "remote_list_dp": 50,
        "remote_select": 1,
        "reset_t": 5,
        "rf_width": 350,
        "rl1_lv": 1,
        "rl1_pin": 6,
        "series_ctrl": 0,
        "total_bt_lv": 0,
        "total_bt_pin": 7,
        "total_stat": 2,
        "zero_select": 0
    },
    "gw_di": {
        "abi": 0,
        "id": null,
        "swv": "1.2.1",
        "bv": "40.00",
        "pv": "2.2",
        "lpv": "3.4",
        "pk": "keyjnuy4s3kre7m7",
        "firmk": "keyjnuy4s3kre7m7",
        "cadv": "1.0.5",
        "cdv": "1.0.0",
        "dev_swv": "1.2.1",
        "s_id": null,
        "dtp": 0,
        "sync": 0,
        "attr_num": 1,
        "mst_tp_0": 9,
        "mst_ver_0": "1.2.1",
        "mst_tp_1": 0,
        "mst_ver_1": null,
        "mst_tp_2": 0,
        "mst_ver_2": null,
        "mst_tp_3": 0,
        "mst_ver_3": null
    },
    "wd_channel_mem": {
        "ch0": 1
    },
    "gw_wsm": {
        "nc_tp": 1,
        "ssid": null,
        "passwd": null,
        "md": 0,
        "random": 0,
        "wfb64": 1,
        "stat": 0,
        "token": null,
        "region": null,
        "reg_key": null,
        "dns_prio": 0
    },
    "wf_start_md": 3,
    "gw_ai": {
        "key": null,
        "lckey": null,
        "h_url": null,
        "h_ip": null,
        "hs_url": null,
        "hs_ip": null,
        "hs_psk": null,
        "hs_psk_ip": null,
        "mqs_url": null,
        "mqs_ip": null,
        "mq_url": null,
        "mq_ip": null,
        "ai_sp": null,
        "ai_sp_ip": null,
        "mq_psk": null,
        "mq_psk_ip": null,
        "lp_url": null,
        "lp_ip": null,
        "time_z": null,
        "s_time_z": null,
        "wx_app_id": null,
        "wx_uuid": null,
        "dy_tls_m": 0,
        "cloud_cap": 0,
        "psk21_key": null
    }
}

Cossid, 2023-11-08:
We're going to need a full dump for this, as there are more than one firmware for 1.2.1 bk7231n plugs, and we cannot tell which it needs from the data alone.

nookbg, 2023-11-08:
i dont know how to upload  dump.bin..

Cossid, 2023-11-08:
You can zip it and attach it here, or just rename the extension to .zip if necessary.

nookbg, 2023-11-08:
[mk601 mini relay BK7231N_QIO_new_2023-17-10-09-23-46.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13301827/mk601.mini.relay.BK7231N_QIO_new_2023-17-10-09-23-46.zip)

Cossid, 2023-11-08:
Do you have a link for this product?

Cossid, 2023-11-19:
Found it, added as `Tuya Generic MK601 Mini Relay`

closed 2023-11-19 by Cossid