tuya-cloudcutter/tuya-cloudcutter #525: LSC 3200719 Not Working

drschnalli, 2023-11-10:
Hey there, just got a new LSC Outdoor RGB Floodlight. It Got Main and MCU Firmware V1.3.22 (which is not available in selection) only 1.3.21 and 1.3. How can i get this working?

Cossid, 2023-11-10:
Can you clarify or post logs of what you actually did?  I find it unlikely that the 1.3.21 profile worked with a device with 1.3.22 listed as the firmware version.  Sending the default response is just for Tuya events we don't need/handle, but it shouldn't have been listed for every request.  But you say you are now on 3rd party firmware, was that via CloudCutter or serial flashing?

If OpenBeken supports a way to dump the Tuya storage of the device, could you post it here?

drschnalli, 2023-11-10:
Sure. 

I try to get the whole bash history. And will also post it here. 

What i did was choosing 3rd Party Firmware and i think i searched by firmware and used 1.3.21 and at the end i've chosen OpenBeken-v1.17.262_bk7231n and this time it worked. 

[BK7231N_TuyaConfig_floodlight.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13321095/BK7231N_TuyaConfig_floodlight.txt)

Cossid, 2023-11-10:
Well, this is indeed rare, and apparently the first time this has happened, where the same offsets work against two different firmware versions.  But to be fair, these are very close version numbers of the same type of firmware, and I'm guessing they didn't make any code changes, likely only text resource changes, or the changes were so minor there were no conflicting calls at the addresses.

I have added a new profile as `LSC 3200719 Outdoor Flood Light`

closed 2023-11-10 by Cossid

drschnalli, 2023-11-11:
Just one more question... am i supposed to use openbeken or can i also switch to tasmota/esphome aswell?

Cossid, 2023-11-11:
Tasmota does not support Beken devices, but you can change to ESPHome as well; https://docs.libretiny.eu/docs/flashing/esphome/#migrating-from-esphome-to-openbeken

drschnalli, 2023-12-02:
@Cossid am i able to directly flash esphome via openbeken?