tuya-cloudcutter/tuya-cloudcutter #527: LVL ceiling light with firmware 1.3.2

zebrahosting, 2023-11-11:
Trying to find the right firmware for my ceiling lights. I do have many different sizes and they are branded LVL and all run firmware 1.3.2
They have only a color temp setting, no RGB and dimming
Anybody successfully flashed those?
[(https://www.aliexpress.com/item/1005005201427570.html?spm=a2g0o.productlist.main.13.389fF5MAF5MAG1&algo_pvid=79af3740-2001-4d28-b641-6eea58dd453e&algo_exp_id=79af3740-2001-4d28-b641-6eea58dd453e-6&pdp_npi=4%40dis%21EUR%2178.23%2124.25%21%21%21594.94%21%21%402101e56616997045835196899e3319%2112000032129130147%21sea%21TH%21783840465%21&curPageLogUid=H8bRR5F3WHBh)]
There are many lookalikes so probably rebranded.

Cossid, 2023-11-11:
There is an existing profile for ceiling lights with firmware 1.3.2, it should work fine.  Choose by firmware version/name and select `1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty`.

closed 2023-11-11 by Cossid

zebrahosting, 2023-11-12:
OK tried this version: 
Got through the first steps but ended in error when finishing:

Using WLAN adapter: wlan0
Nov 12 08:47:56 dnsmasq[15]: started, version 2.80 cachesize 150
Nov 12 08:47:56 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Nov 12 08:47:56 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Nov 12 08:47:56 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Nov 12 08:47:56 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:56:c9:51 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Nov 12 08:48:01 dnsmasq-dhcp[15]: 2058546371 available DHCP range: 10.42.42.10 -- 10.42.42.40
Nov 12 08:48:01 dnsmasq-dhcp[15]: 2058546371 client provides name: wlan0
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 475, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 239, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmware=True)
  File "/src/cloudcutter/__main__.py", line 96, in __configure_local_device_or_update_firmware
    authkey, uuid, pskkey = config.get_bytes(DeviceConfig.AUTH_KEY, default=DEFAULT_AUTH_KEY), config.get_bytes(DeviceConfig.UUID), config.get_bytes(DeviceConfig.PSK, default=None)
  File "/src/cloudcutter/device.py", line 29, in get_bytes
    return self.get(key, default).encode(encoding)
AttributeError: 'NoneType' object has no attribute 'encode'
Oh no, something went wrong with updating firmware! Try again I guess...

Several attempts give slightly different results. I have seen a Cloutcutter AP name poping up once but now it is back to the original one. I can still get it in pairing mode.

I run it also with the -v flag but dont see much more output. Does it store a logfile somewhere?