tuya-cloudcutter/tuya-cloudcutter #528: AttributeError: 'NoneType' object has no attribute 'encode'

szupi-ipuzs, 2023-11-11:
I am trying to flash esphome to my circuit breaker, but cloudcutter failed as below.
Could this be related to the todays changes of @Cossid? I will try earlier version and report back.

```
Using WLAN adapter: wlx647002177bee

Configuration file: /dev/stdin

Using interface wlx647002177bee with hwaddr 64:70:02:17:7b:ee and ssid "cloudcutterflash"

wlx647002177bee: interface state UNINITIALIZED->ENABLED

wlx647002177bee: AP-ENABLED 

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main

    return _run_code(code, main_globals, None,

  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code

    exec(code, run_globals)

  File "/src/cloudcutter/__main__.py", line 475, in <module>

    args.handler(args)

  File "/src/cloudcutter/__main__.py", line 239, in __update_firmware

    __configure_local_device_or_update_firmware(args, update_firmware=True)

  File "/src/cloudcutter/__main__.py", line 96, in __configure_local_device_or_update_firmware

    authkey, uuid, pskkey = config.get_bytes(DeviceConfig.AUTH_KEY, default=DEFAULT_AUTH_KEY), config.get_bytes(DeviceConfig.UUID), config.get_bytes(DeviceConfig.PSK, default=None)

  File "/src/cloudcutter/device.py", line 29, in get_bytes

    return self.get(key, default).encode(encoding)

AttributeError: 'NoneType' object has no attribute 'encode'

Oh no, something went wrong with updating firmware! Try again I guess...

If you need to report an issue, please run with the -v flag and supply the full log of that attempt.
```

The command line was:
```
sudo ./tuya-cloudcutter.sh -p tuya-generic-zmb9-1p-circuit-breaker -f ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
```

Cossid, 2023-11-11:
Yeah, looks like I probably broke something.  Give me a sec to look into what the issue is.

Cossid, 2023-11-11:
@szupi-ipuzs Pushed a fix, can you please give it a try?

szupi-ipuzs, 2023-11-12:
Thank you for your fix, although I have not anticipated it so fast and... I reverted to older code and it worked which means I have it on kickstarter now.
Don't have any other devices to try right now.

Cossid, 2023-11-12:
No worries, I just tested it out myself, and it is working correct now.

closed 2023-11-12 by Cossid