tuya-cloudcutter/tuya-cloudcutter #529: LVL ceiling light firmware 1.3.2

zebrahosting, 2023-11-12:
OK tried the version: 1.3.2 for ceiling light as mentioned in the closed issue before.
Got through the first steps but ended in error when finishing:

Using WLAN adapter: wlan0
Nov 12 08:47:56 dnsmasq[15]: started, version 2.80 cachesize 150
Nov 12 08:47:56 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Nov 12 08:47:56 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Nov 12 08:47:56 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Nov 12 08:47:56 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:56:c9:51 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Nov 12 08:48:01 dnsmasq-dhcp[15]: 2058546371 available DHCP range: 10.42.42.10 -- 10.42.42.40
Nov 12 08:48:01 dnsmasq-dhcp[15]: 2058546371 client provides name: wlan0
Traceback (most recent call last):
File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
return _run_code(code, main_globals, None,
File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
exec(code, run_globals)
File "/src/cloudcutter/main.py", line 475, in
args.handler(args)
File "/src/cloudcutter/main.py", line 239, in __update_firmware
__configure_local_device_or_update_firmware(args, update_firmware=True)
File "/src/cloudcutter/main.py", line 96, in __configure_local_device_or_update_firmware
authkey, uuid, pskkey = config.get_bytes(DeviceConfig.AUTH_KEY, default=DEFAULT_AUTH_KEY), config.get_bytes(DeviceConfig.UUID), config.get_bytes(DeviceConfig.PSK, default=None)
File "/src/cloudcutter/device.py", line 29, in get_bytes
return self.get(key, default).encode(encoding)
AttributeError: 'NoneType' object has no attribute 'encode'
Oh no, something went wrong with updating firmware! Try again I guess...

Several attempts give slightly different results. I have seen a Cloutcutter AP name poping up once but now it is back to the original one. I can still get it in pairing mode.

I run it also with the -v flag but dont see much more output. Does it store a logfile somewhere?

Cossid, 2023-11-12:
Please do a `git pull` to update to the latest code, this was fixed yesterday.

zebrahosting, 2023-11-13:
Using WLAN adapter: wlan0
Nov 13 07:35:05 dnsmasq[15]: started, version 2.80 cachesize 150
Nov 13 07:35:05 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Nov 13 07:35:05 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Nov 13 07:35:05 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Nov 13 07:35:05 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:56:c9:51 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Nov 13 07:35:10 dnsmasq-dhcp[15]: 1864448219 available DHCP range: 10.42.42.10 -- 10.42.42.40
Nov 13 07:35:10 dnsmasq-dhcp[15]: 1864448219 client provides name: wlan0
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 475, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 239, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmware=True)
  File "/src/cloudcutter/__main__.py", line 96, in __configure_local_device_or_update_firmware
    authkey, uuid, pskkey = config.get_bytes(DeviceConfig.AUTH_KEY, default=DEFAULT_AUTH_KEY), config.get_bytes(DeviceConfig.UUID), config.get_bytes(DeviceConfig.PSK, default=None)
  File "/src/cloudcutter/device.py", line 29, in get_bytes
    return self.get(key, default).encode(encoding)
AttributeError: 'NoneType' object has no attribute 'encode'
Oh no, something went wrong with updating firmware! Try again I guess...
===================

I did a git pull and started over.

When I start the light it immediatly show an access point so no need to make it glow.
During the process it changes a few times from glowing to full etc.

After the first upload, it finds the light straight away, it seems it gets a little further than yesterday.

Cossid, 2023-11-13:
The log you posted indicates you didn't actually update, that isn't the latest code.

zebrahosting, 2023-11-14:
OK delete the PI image and started over. FInally got it flashed and it popped up in Home assistant but adoption was not possible because API was missing from the YAML it said. Surprisingly after a few hours it did adopt without issue but is now asking for pins... No idea how or where to find the info. It's currently using FW Kickstart-Bk7231T and I wonder how to get the brightness and colortemp controls back.
Thanks for the cloudcutter work, one done, 12 more to go....

Cossid, 2023-11-14:
The api encryption key is something that had to have been in your config, there is nothing we can do to help you there if you used your own firmware.  Kickstart doesn't support device functionality, it is merely meant to be used as an easy stepping stone (it also doesn't have an API encryption key).

closed 2023-11-14 by Cossid