tuya-cloudcutter/tuya-cloudcutter #531: Generic UK plug WHDZ03 -- BK7231N firmware 1.1.17 not exploitable

ben-willmore, 2023-11-15:
I found that the above plug could not be exploited using the profile for 1.1.17 firmware.

On dumping the flash, I found the following:

```
[+] Matched pattern for BK7231N version SDK 2.3.3 LAN 3.4/CAD 1.0.5, payload type ssid
[!] The binary supplied appears to be patched and no longer vulnerable to the tuya-cloudcutter exploit.
```

I can provide the dump if helpful. Also I thought it might be useful to update the 'known patched' page.

Cossid, 2023-11-15:
Added to the known patched list.  If you don't mind, go ahead and post the dump anyway.  CloudCutter won't be able to support it, but we're still collecting info about newer builds that may be exploitable in the future.

closed 2023-11-15 by Cossid

ben-willmore, 2023-11-15:
[Tuya-Generic_WHD---Z03.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13369113/Tuya-Generic_WHD---Z03.bin.zip)