tuya-cloudcutter/tuya-cloudcutter #532: Installation issue

nhaase23, 2023-11-19:
Hi,

im fairly new here and forgive me, if im doing something wrong. So when i run the tuya-cloudcutter.sh command in Ubuntu 22.04 i get the following issue. Am i doing something wrong?

```
[!] WARNING: Selected wifi AP support: ja
AP support is manditory for tuya-cloudcutter to work.  If this is blank or 'no' your adapter doesn't support this feature.
Press any key to continue, or CTRL+C to exitBuilding cloudcutter docker image
[+] Building 3.7s (10/12)                                        docker:default
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 460B                                       0.0s
 => [internal] load metadata for docker.io/library/python:3.9.12-slim-bus  1.1s
 => [internal] load build context                                          0.0s
 => => transferring context: 3.02kB                                        0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.12-slim-buster@sha256:83  0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                   0.0s
 => ERROR [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipen  2.4s
------                                                                          
 > [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy:                                                                           
0.940 Creating a virtualenv for this project...                                 
0.941 Pipfile: /src/Pipfile                                                     
0.955 Using /usr/local/bin/python3.9 (3.9.12) to create virtualenv...           
2.080 created virtual environment CPython3.9.12.final.0-64 in 810ms
2.080   creator CPython3Posix(dest=/src/.venv, clear=False, no_vcs_ignore=False, global=False)
2.080   seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
2.080     added seed packages: pip==23.3.1, setuptools==68.2.2, wheel==0.41.2
2.080   activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
2.080 
2.081 ✔ Successfully created virtual environment!
2.142 Virtualenv location: /src/.venv
2.198 Your Pipfile.lock (b1a6f3) is out of date.  Expected: (019fbf).
2.207 Usage: pipenv install [OPTIONS] [PACKAGES]...
2.207 
2.207 ERROR:: Aborting deploy
------
Dockerfile:11
--------------------
   9 |     COPY src/Pipfile /src/
  10 |     COPY src/Pipfile.lock /src/
  11 | >>> RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
  12 |     
  13 |     FROM python-deps AS cloudcutter
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy" did not complete successfully: exit code: 2
Failed to build Docker image, stopping script
```

ghostwhowalkz, 2023-11-19:
Same issue here, running on the suggested version of Raspbian... Been working through methods of updating the piplock file, but nothing working so far.

nhaase23, 2023-11-19:
Glad to here, that im not the only one. Have you managed to get it working?

Cossid, 2023-11-19:
> 2.198 Your Pipfile.lock (b1a6f3) is out of date.  Expected: (019fbf).

`019fbf` is what is currently checked in, it was updated yesterday.  Please do a `git pull` and a `docker buildx purge` and try again?

Cossid, 2023-11-19:
Well, looks like one line of the Pipfile hadn't updated correct, commited again, please update and try again.

nhaase23, 2023-11-19:
Yes its working now, tank you!

closed 2023-11-19 by nhaase23