tuya-cloudcutter/tuya-cloudcutter #534: Busted dnsmasq version - update image

duk3luk3, 2023-11-21:
The dnsmasq command in setup_apmode.sh is hanging for some reason.

Here is the config the script is running with:

```
run_in_docker bash -c bash /src/setup_apmode.sh wlp2s0 true && pipenv run python3 -m cloudcutter update_firmware "/work/device-profiles/arlec-pc191ha-smart-plug-bk7231t-v1.3.5/combined.json" "/work/device-profiles/schema" "/work/configured-devices/327rHZmmdpGc.deviceconfig" "/work/custom-firmware/" "ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin" "15" "true"
```

And here is the dnsmasq command it runs:

```
dnsmasq --conf-file=/dev/null --no-resolv --interface=wlp2s0 --bind-interfaces --listen-address=10.42.42.1 --except-interface=lo --dhcp-range=10.42.42.10,10.42.42.40,12h --address=/#/10.42.42.1 -x /src/dnsmasq.pid --log-dhcp --log-queries --log-facility=/dev/stdout
```

dnsmasq runs but it just hangs - it doesn't daemonize itself.

When I strace it, it is looping on `close(xxxxxxx) -1 EBADF`, I believe this is because of an old dnsmasq bug: https://bugzilla.redhat.com/show_bug.cgi?id=1816613

After changing the docker base image to `python:3.9.18-slim-bullseye`, the problem no longer occurs.

Cossid, 2023-11-21:
Interesting.  What are the indicators it is hanging?  We pinned to 3.9.12-buster because our base was already buster, and after 3.9.12 they stopped supporting armv5/6 which some RPIs run (rpi 1, zero lines).  I see the 3.9 bullseye line, up to the latest 3.9.18 still support armv5/6.

I'll quickly test 3.9.18-bullseye on rpi4 myself, and if it works, change to that.  Anybody who needs older we can simply instruct to change a single line.

Cossid, 2023-11-21:
Changed default to 3.9.18-bullseye.

closed 2023-11-21 by Cossid