tuya-cloudcutter/tuya-cloudcutter #537: Please create new device profile - Zemismart TUYA WiFi Smart LED Ceiling Lamp 34W

MnM001, 2023-11-22:
HI,

I have this light:

Zemismart TUYA WiFi Smart LED Ceiling Lamp 34W

It shows firmware v1.0.0 -  BK7231N

Attached is the firmware for it.

[firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/files/13441584/firmware.zip)

Many thanks.

Cossid, 2023-11-22:
This doesn't appear to be a standard Tuya firmware, the app partition is at a different offset, and it doesn't seem to be CRC16 compliant.  Not quite sure what all of that means yet.  The bootloader and storage areas appear to be valid.  I'll look into it more as time allows.

MnM001, 2023-11-22:
[This is the teardown for it.](https://www.elektroda.com/rtvforum/topic4002073.html)

Currently running ESPHome on the device without any issues.

Cossid, 2023-12-06:
@openshwprojects This has now become a pattern of users who apparently tried dumping firmware with BK7231 Easy Flasher and would up with corruption from 0x11000 to 0x13000.  This one has a data chuck repeated, where the other just has all 0xFF throughout.

This was the first case, the other issue being #554 (where it was confirmed that the corruption was actually written to the flash).

Closing, as this is just a corrupt image we can't correct for.

closed 2023-12-06 by Cossid

Cossid, 2023-12-06:
@openshwprojects apparently this is the result of the advanced option `Test read/write`.  I would suggest giving a warning that it will corrupt the existing image at the very least.  Possibly change your test write address to OTA partition as well.

openshwprojects, 2023-12-07:
@Cossid  thanks for pointing that out, I would never expect that someone would wander to advanced options to test write before doing backup. I'll disabled it for now. Btw, if you are still looking for flash dumps, you can check here: https://github.com/openshwprojects/FlashDumps