tuya-cloudcutter/tuya-cloudcutter #538: how to "verify" if the device is bricked?

gytisgreitai, 2023-11-22:
Hey,

thanks for the scripts, everything seemed to be ok, until I decided to flash the `ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin` firmware (Generic tuya LSPA9 Plug v1.0.3)

Flashing hanged, the light is pale blue, does not react to any presses, does not connect to the wifi. So I assume that I've bricked it, and the only way is to dismantle it, solder up the serial, and try going from there?

```
Using PSK v1 - Received PSK ID version 01
Processing endpoint /device/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.upgrade.get
Processing endpoint tuya.device.upgrade.status.update
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
Firmware update progress: 15%
Processing endpoint tuya.device.uuid.pskkey.get
Firmware update progress: 30%
Firmware update progress: 60%
Firmware update progress: 69%
Firmware update progress: 90%
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin send complete, request range: bytes=0-438751/438752
Processing endpoint atop.online.debug.log
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dynamic.config.ack
```

Cossid, 2023-11-22:
ESPHome Kickstart does not connect to your wifi, it starts in AP mode.  But it is strange you didn't get a 98% firmware update progress, usually the last one we get.

Kickstart firmware also has no device functionality, so buttons are not expected to work.

gytisgreitai, 2023-11-22:
uh, sorry I guess I missed the readme somewhere :( And that hanged script got me confused and I made wrong assumptions.

All working, AP mode available.

Thanks again!

closed 2023-11-22 by gytisgreitai

OZub, 2024-03-20:
so how did you fix it?
 I got the same issue, mine is stuck at 30%, request range: bytes=0-562447/562448