tuya-cloudcutter/tuya-cloudcutter #539: What am I doing wrong?

maxmanager, 2023-11-23:
I have 7 smart plugs from different manufacturers and 1 smart switch from Moes. Their firmware is 1.1.7 or 1.1.8. I tried them all using the firmware and in the case of Moes I tried his profile. I tried with RPI 3 and 4 and also on my PC with Ubuntu. All my attempts always give exactly the same result: "The profile you selected did not result in a successful exploit.". I don't know what else to do, could you tell me what I'm doing wrong?

[image]

maxmanager, 2023-11-23:
I also tried using Lightleak on 2 plugs and it even connected but when I tried to Read Flash the app gave me an error message.

Cossid, 2023-11-24:
In the case of the Moes, that profile is for a relay, and there is no guarantee they only have one model or one firmware.  Did you check to see if yours has a matching 1.1.6 firmware?

For the others, are you certain you have Beken devices?  1.1.8 is a common plug firmware and should exploit if you use the correct 1.1.8 profile, but there is never a guarantee.  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out might help verify if you have an ESP module rather than beken.

maxmanager, 2023-11-24:
this is a 1.1.7 which is an esp

[image]

Cossid, 2023-11-24:
ESP devices are not supported by Tuya-CloudCutter, only Beken devices currently.

closed 2023-11-28 by Cossid