tuya-cloudcutter/tuya-cloudcutter #541: Dump for SCW_SL_CBHC3-108 Generic Smart led RGB strip bars

andreq, 2023-11-25:
Made a dump using ltchiptool and removed my SSID and pwd using an hex editor (hopefully this isn't a problem)
Chip is a BK7231N

Dump :
[Uploading ltchiptool_bk72xx_2023-11-24_22-08-54_NO_SSID.zip…]()

Image of the device : 
[image]
[image]

Currently trying to find the pinout blindly as ltchiptool wasn't able from the dump or esp-home kickstart.

It has 
3 button
1 IR receiver
and what look's to be an addressable RGB data out IO that connect to a secondary board 
[image]
(this is not 12v, I think the mfg re-use the same splitter board for many device)

It's this thing on aliexpress : https://s.click.aliexpress.com/e/_De8V3Lp

Cossid, 2023-11-25:
Your dump didn't properly attach.

andreq, 2023-11-25:
[ltchiptool_bk72xx_2023-11-24_22-08-54_NO_SSID.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13462978/ltchiptool_bk72xx_2023-11-24_22-08-54_NO_SSID.zip)

Hope this works

Cossid, 2023-11-25:
Added as `Tuya Generic SCW-SL-CBHC3-108 Decorative Wall Lights`.  As expected, no additional pin configuration available.  Looks like this is RGBIC, so currently no support from ESPHome until SPI is integrated, or NeoPixelBus is modified to support LibreTiny.  Unsure about OpenBeken support.

closed 2023-11-25 by Cossid

andreq, 2023-11-26:
I'm having some kind of success with OpenBeken SM16703 support, but I think the leds are WS2812b so it's not exactly there yet.

Thanks for the quick addition.