tuya-cloudcutter/tuya-cloudcutter #542: New profile needed for Quanquer Smart Christmas Lights Using (CZJUTAI Tuya Wifi controller)

mppkll, 2023-11-26:
I found that the firmware is v1.0.8 and the board is a CB2S with a BK7231N chip.  Firmware full dump attached (mfg default WiFi/config and never connected to Tuya SmartLife app at the time of the dump).

[CZJUTAI-multi-lights-FLASH.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13465780/CZJUTAI-multi-lights-FLASH.zip)

[image]
[image]

Cossid, 2023-11-26:
Added as `Quanquer CZJUTAI String Lights`.  Looks like the detected device_configuration/user_primary_key is not from storage, and likely belongs to a sample light configuration that doesn't truly belong to this device.  I'm not sure if this device is regular RGB or if it is RGBIC, in which case, there will be limited 3rd party firmware support.

closed 2023-11-26 by Cossid

mppkll, 2023-11-26:
> Added as `Quanquer CZJUTAI String Lights`. Looks like the detected device_configuration/user_primary_key is not from storage, and likely belongs to a sample light configuration that doesn't truly [...]

I found that It is RGBIC and got more info on this site (https://cpldcpu.wordpress.com/2022/01/23/controlling-rgb-leds-with-only-the-powerlines-anatomy-of-a-christmas-light-string/) as to how the communication protocol works for the LED string.  I tested the light string and found that it is the same protocol as discussed in the above site. The circuit for the controller has a unmarked 8 pin IC (MCU?) that is controlling the power MOSFETs and sending the pulses to the LED string, so I don't think that part will be a worry since something is already generating the needed pulses.  On the CB2S, P6 is connected to a status LED, P7 is connected to the button, and P11/TX1 is connected to the unmarked IC.  Looking at the signal from the unmarked IC I think it might be I2C comms.  I'm going to try decoding it.