tuya-cloudcutter/tuya-cloudcutter #543: Compatibility Inquiry for eCosy Device Sold by Konyks

matguig, 2023-11-27:
Hello Tuya Cloudcutter Team,

I am reaching out to inquire about the compatibility of a specific device with your project. The device in question is the "[eCosy](https://konyks.com/produit/ecosy/)" smart home product, which is marketed by Konyks. I have recently been exploring the capabilities of Tuya CloudCutter and am interested in integrating my eCosy device.

I noticed that the eCosy device is not listed in the compatibility list. I am curious to know if there is a possibility of this device being compatible with tuya-cloudcutter, or if there are any ongoing developments towards supporting it.

Thank you for your time and assistance.

Best regards,
Matthieu

Cossid, 2023-11-27:
Depends on what module is in the device, and which firmware it has.  If you're unsure of the module, you may have to open the device an look.  To find out the firmware version, see https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

If there is not a compatible firmware currently, a serial dump would be needed to support the device.

matguig, 2023-11-27:
> Depends on what module is in the device, and which firmware it has. If you're unsure of the module, you may have to open the device an look

As I understand it, the module is TYWE3S.
The firmware version is v1.0.9

[image]
[image]