tuya-cloudcutter/tuya-cloudcutter #546: Accidently flashed ESPHome and not BK7231N durning inital setup

NielsPiersma, 2023-11-30:
I understand this request doesn't belong here, but I can't find a solution for accidentally flashing the wrong version (ESPHome vs BK7231N) on a device during cloudcutter.

Could someone give me a hand on how to reflash (a filament bulb in this case) with the BK7321n firmware.

Already tried uploading via OTA, but it expects an uf2 file, and I can't find a tool for converting rbl to uf2...

Again, sorry, I don't know this is not a tuya-cloudcutter issue, but any help/hints are appreciated.

niels

Cossid, 2023-11-30:
You appear to be using incorrect terms here.  BK7231N is the chip type, ESPHome is a firmware (which can be compiled for BK7231T or BK7231N, but CloudCutter will only allow you to flash the chip type of the profile you are using without extra wrok).  Are you trying to say you meant to flash OpenBeken instead of ESPHome, or are you saying you have a BK7231T device and flashed BK7231N firmware?

NielsPiersma, 2023-11-30:
Cossid,

You are right, and I am sorry for the misunderstanding.

I've been flashing BK7231T/N chips using cloudcutter for a long time. And that was never an issue.
Start cloudcutter, stop DNS, select the device and select the firmware.

In this install run, I mistakenly chose the ESPHome Kickstart firmware and not the "normal" rbl file 

[image]

So the flashing went fine, and I can access the device, but it is running ESPHome Kickstart and not OpenBeken v1.17.230.

Yes, certainly my mistake, but I am looking for a solution to undo my mistake ;)

Niels

closed 2023-11-30 by Cossid