tuya-cloudcutter/tuya-cloudcutter #549: Ledvance SmartPlus Plug EU not working

besanur, 2023-12-02:
I'm trying to flash ESP Kickstarter onto the Ledvance Plug. But I haven't had any success so far. Here https://upk.libretiny.eu/?profile=ledvance-smartplus-plug-eu the slug says **bk7231s** but on this page it says **BK7231t** as the chip https://templates.blakadder.com/ledvance_4058075522800.html . Are there multiple versions?

For me I see the firmware version 1.1.7, which doesn't match the profile on https://upk.libretiny.eu/ either.

I always end up with 

> [!] The profile you selected did not result in a successful exploit.

```bash
pi@laborpi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh 
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Building cloudcutter docker image
[+] Building 4.8s (13/13) FINISHED                                                                                                                                                       docker:default
 => [internal] load .dockerignore                                                                                                                                                                  0.1s
 => => transferring context: 2B                                                                                                                                                                    0.0s
 => [internal] load build definition from Dockerfile                                                                                                                                               0.0s
 => => transferring dockerfile: 462B                                                                                                                                                               0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                                                                                     1.5s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5505ae04f1                                                          1.3s
 => => resolve docker.io/library/python:3.9.18-slim-bullseye@sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5505ae04f1                                                               1.3s
 => [internal] load build context                                                                                                                                                                  0.5s
 => => transferring context: 3.02kB                                                                                                                                                                0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                               0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                                      0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                                                0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                                           0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                                       0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                                         0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                                          0.0s
 => exporting to image                                                                                                                                                                             0.0s
 => => exporting layers                                                                                                                                                                            0.0s
 => => writing image sha256:e30bce802a099b24532f6669ec99f8aa7a86c1ae775f1e02c780dcfee0bdd1bb                                                                                                       0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                                                     0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.7 - BK7231N / oem_bk7231n_plug
   1.1.6 - BK7231T / oem_bk7231s_light_ffc_ir_key_td
   1.1.6 - BK7231T / oem_bk7231s_light_ty
   1.1.6 - BK7231T / oem_bk7231s_strip_ty
   1.1.6 - BK7231T / oem_bk7231s_switch1_zero_kjd
   1.1.7 - BK7231N / CL_DREAM_RGB_STRIP_19KEY_1KEY_5V_BK7231N
   1.1.7 - BK7231N / oem_bk7231n_ceiling_light_ty
 > 1.1.7 - BK7231N / oem_bk7231n_plug
   1.1.7 - BK7231T / oem_bk7231s_light_pir_ty
   1.1.7 - BK7231T / oem_bk7231s_light_ty
   1.1.71 - BK7231T / bk7231t_common_user_config_ty
   1.1.8 - BK7231N / farylink_soc_hzs_5_light_15155
   1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231n.ug.bin

Selected Device Slug: lsc-2578685-970766-smart-plug-cb2s
Selected Profile: oem-bk7231n-plug-1.1.7-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
..
Found access point name: "LDV SMART+ -9534", trying to connect...
Device 'wlan0' successfully activated with '46300c30-64a9-4b31-8779-dacecfcf326e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/hHNpWeZkahcq.deviceconfig
Saved device config in /work/configured-devices/hHNpWeZkahcq.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "LDV SMART+ -9534", trying to connect...
Device 'wlan0' successfully activated with '46300c30-64a9-4b31-8779-dacecfcf326e'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

kuba2k2, 2023-12-02:
You selected BK7231N and that's not going to work. BK7231S is the same as BK7231T. Try one of the T profiles instead. If there isn't a working plug profile for your device, you'll have to dump the firmware by serial.

Cossid, 2023-12-02:
As kuba2k2 mentioned, Tuya sometimes uses BK7231S internally in their SDK to reference BK7231T, they are the same.

The only firmware we have for this device is 7.0.0, and there is no update available for it.  Are you sure you don't have an older ESP version of the plug?  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

There are no BK7231T profiles for 1.1.7 plug, and since the BK7231N profile didn't work, your plug wasn't swapped to a BK7231N module.

Either your device is not Beken, or we would need a new full dump to support it.

besanur, 2023-12-02:
> You selected BK7231N and that's not going to work. BK7231S is the same as BK7231T. Try one of the T profiles instead. If there isn't a working plug profile for your device, you'll have to dump the firmware by serial.

thanks for the quick reply. I tried several and of course also BK7231T. I think it's probably the ESP variant, as @Cossid already mentioned. Thank you both.