tuya-cloudcutter/tuya-cloudcutter #550: New Device Profile - Costco - Feit Smart Color Chasing Strip Light

feene, 2023-12-02:
Believe it is 1.0.2 for Module and MCU.
Hoping for a device profile.
New LED Strip Available at Costco now.
FETAPE20/CONT

[Costco Feit FETAPE20-CONT.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13537076/Costco.Feit.FETAPE20-CONT.zip)

[image]

Cossid, 2023-12-02:
This device is patched and serial flashing will be your only option.  Being RGBIC, it's likely SPI and you'll have limited support for that portion with 3rd party firmware.

Here is the user_param_key section:

```json
{
	"CS": 15,
	"Jsonver": "1.0.4",
	"LedNum": 48,
	"SCL": 14,
	"adclimit": 2400,
	"aging": 1,
	"brifollow": 1,
	"brightmax": 58,
	"brightmin": 10,
	"brightstep": 20,
	"c_lv": 1,
	"c_pin": 8,
	"cagt": 20,
	"category": "0505",
	"cmod": "rgbcw",
	"colormax": 100,
	"colormaxp": 100,
	"colormin": 10,
	"colororder": 0,
	"colorpfun": 1,
	"crc": 71,
	"ctrl_lv": 1,
	"ctrl_pin": 22,
	"defbright": 100,
	"defcolor": "c",
	"deftemp": 0,
	"gmkb": 55,
	"gmkg": 60,
	"gmkr": 100,
	"gmwb": 35,
	"gmwg": 40,
	"gmwr": 100,
	"irfunc": 0,
	"ismusic": 0,
	"k1dfunc": 4,
	"key_lv": 0,
	"key_pin": 24,
	"keyfunc": 1,
	"knum": 1,
	"ktime": 3,
	"micpin": 23,
	"micproc": 800,
	"miso": 17,
	"module": "CBU",
	"mosi": 16,
	"onoffmode": 0,
	"pairt": 180,
	"pmemory": 1,
	"prodagain": 0,
	"pwmhz": 1000,
	"remdmode": 1,
	"rgbt": 10,
	"rstbr": 50,
	"rstcor": "c",
	"rstmode": 1,
	"rsttemp": 100,
	"sensimax": 300,
	"sensimin": 100,
	"sfunc": 1,
	"slidemod": 2,
	"title20": 0,
	"totallen": 1200,
	"w_lv": 1,
	"w_pin": 7,
	"wfcfg": "spcl",
	"wfct": 3,
	"wt": 20
}
```

Full storage with DPID schema (which may have some remote code info) is
```json
{
	"gw_bi": {
		"uuid": "322aacd432317060",
		"psk_key": "dZ7bFzvEPQXKnfozBPuu6yQgDYtcXl7STr77X",
		"auth_key": "wePPvPpmFYFla1RA7YyWZQkAOmw6iaZR",
		"ap_ssid": "SmartLife",
		"ap_passwd": null,
		"country_code": "CN",
		"bt_mac": null,
		"bt_hid": null,
		"prod_test": false,
		"fac_pin": "expmpw4xxd0kkifb"
	},
	"gw_di": {
		"abi": 0,
		"id": "ebba4530f41c44cd94bbt6",
		"swv": "1.0.2",
		"bv": "40.00",
		"pv": "2.2",
		"lpv": "3.4",
		"pk": "keyhj9djx7whse3h",
		"firmk": "keyhj9djx7whse3h",
		"cadv": "1.0.5",
		"cdv": "1.0.0",
		"dev_swv": "1.0.2",
		"s_id": "e6hr30",
		"dtp": 0,
		"sync": 0,
		"attr_num": 1,
		"mst_tp_0": 9,
		"mst_ver_0": "1.0.2",
		"mst_md5_0": null,
		"mst_tp_1": 0,
		"mst_ver_1": null,
		"mst_md5_1": null,
		"mst_tp_2": 0,
		"mst_ver_2": null,
		"mst_md5_2": null,
		"mst_tp_3": 0,
		"mst_ver_3": null,
		"mst_md5_3": null
	},
	"tls_ca_cnt": 0,
	"rmt_ctrl_0": {
		"no": 1,
		"mac": "3CNPngKh",
		"grpid": 2,
		"s2": "v7rqfvHr8x9LjtMU4Qvi+g==",
		"sn": 204,
		"cld": 1,
		"key_tp": 0,
		"s1": "pKE="
	},
	"rmt_ctrl_2": {
		"no": 3,
		"mac": "3CNPnIKg",
		"grpid": 2,
		"s2": "b4T6jHldMkB8JxSZlCbC+w==",
		"sn": 212,
		"cld": 1,
		"key_tp": 0,
		"s1": "7aA="
	},
	"gw_wsm": {
		"nc_tp": 9,
		"ssid": "ZmVlbmUxSU9U",
		"passwd": "ZmVlbmUxaW90cXdlYXNk",
		"md": 0,
		"random": 0,
		"wfb64": 1,
		"stat": 2,
		"token": "usKN1mof",
		"region": "AZ",
		"reg_key": "33Mg",
		"dns_prio": 0
	},
	"wf_start_md": 3,
	"rmt_ctrl_1": {
		"no": 2,
		"mac": "3CNPnh4z",
		"grpid": 2,
		"s2": "aeMRFi8ZispIVeATvF2KVA==",
		"sn": 201,
		"cld": 1,
		"key_tp": 0,
		"s1": "HzM="
	},
	"is_stride": 0,
	"ble_beaconkey": "DE564585F7F84349096414B732485157",
	"gw_ai": {
		"key": "y??*6u1H9.SEyf%[",
		"lckey": "~:8:'<:aOayYg|O~",
		"h_url": "http://a.tuyaus.com/d.json",
		"h_ip": "44.241.46.212",
		"hs_url": null,
		"hs_ip": null,
		"hs_psk": "https://a3.tuyaus.com/d.json",
		"hs_psk_ip": "54.69.37.156",
		"mqs_url": null,
		"mqs_ip": null,
		"mq_url": "m2.tuyaus.com:1883",
		"mq_ip": "52.12.196.80",
		"ai_sp": null,
		"ai_sp_ip": null,
		"mq_psk": "m2.tuyaus.com:8886",
		"mq_psk_ip": "52.12.196.80",
		"lp_url": "baal.tuyaus.com",
		"lp_ip": "44.239.29.247",
		"time_z": "-07:00",
		"s_time_z": "[[1710061200,1730620800],[1741510800,1762070400]]",
		"wx_app_id": null,
		"wx_uuid": null,
		"dy_tls_m": 2,
		"cloud_cap": 1025,
		"psk21_key": null
	},
	"e6hr30": [
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 20
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"range": [
					"white",
					"colour",
					"scene",
					"music"
				],
				"type": "enum"
			},
			"id": 21
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 10,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 22
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 23
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 24
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 86400,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 26
		},
		{
			"type": "obj",
			"mode": "wr",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 27
		},
		{
			"mode": "rw",
			"id": 31,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 32,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 33,
			"type": "raw"
		},
		{
			"type": "obj",
			"mode": "ro",
			"property": {
				"min": 1,
				"max": 10000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 46
		},
		{
			"type": "obj",
			"mode": "ro",
			"property": {
				"min": 1,
				"max": 1024,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 47
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 1,
				"max": 1000,
				"scale": 1,
				"step": 1,
				"type": "value"
			},
			"id": 53
		},
		{
			"mode": "rw",
			"id": 56,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 61,
			"type": "raw"
		},
		{
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 71,
			"type": "obj"
		},
		{
			"mode": "rw",
			"id": 210,
			"type": "raw"
		}
	],
	"astro_timer": {
		"timestamp": 1701539485,
		"index": 0
	},
	"mf_test_close": true,
	"em_sys_env": "bk7231n",
	"timer_arr": {
		"lastFetchTime": 0,
		"cnt": 0
	}
}
```

Cossid, 2023-12-02:
This device is patched and serial flashing will be your only option.  Being RGBIC, it's likely SPI and you'll have limited support for that portion with 3rd party firmware.

Here is the user_param_key section:

```json
{
	"CS": 15,
	"Jsonver": "1.0.4",
	"LedNum": 48,
	"SCL": 14,
	"adclimit": 2400,
	"aging": 1,
	"brifollow": 1,
	"brightmax": 58,
	"brightmin": 10,
	"brightstep": 20,
	"c_lv": 1,
	"c_pin": 8,
	"cagt": 20,
	"category": "0505",
	"cmod": "rgbcw",
	"colormax": 100,
	"colormaxp": 100,
	"colormin": 10,
	"colororder": 0,
	"colorpfun": 1,
	"crc": 71,
	"ctrl_lv": 1,
	"ctrl_pin": 22,
	"defbright": 100,
	"defcolor": "c",
	"deftemp": 0,
	"gmkb": 55,
	"gmkg": 60,
	"gmkr": 100,
	"gmwb": 35,
	"gmwg": 40,
	"gmwr": 100,
	"irfunc": 0,
	"ismusic": 0,
	"k1dfunc": 4,
	"key_lv": 0,
	"key_pin": 24,
	"keyfunc": 1,
	"knum": 1,
	"ktime": 3,
	"micpin": 23,
	"micproc": 800,
	"miso": 17,
	"module": "CBU",
	"mosi": 16,
	"onoffmode": 0,
	"pairt": 180,
	"pmemory": 1,
	"prodagain": 0,
	"pwmhz": 1000,
	"remdmode": 1,
	"rgbt": 10,
	"rstbr": 50,
	"rstcor": "c",
	"rstmode": 1,
	"rsttemp": 100,
	"sensimax": 300,
	"sensimin": 100,
	"sfunc": 1,
	"slidemod": 2,
	"title20": 0,
	"totallen": 1200,
	"w_lv": 1,
	"w_pin": 7,
	"wfcfg": "spcl",
	"wfct": 3,
	"wt": 20
}
```

Full storage with DPID schema (which may have some remote code info) is
```json
{
	"gw_bi": {
		"uuid": "322aacd432317060",
		"psk_key": "dZ7bFzvEPQXKnfozBPuu6yQgDYtcXl7STr77X",
		"auth_key": "wePPvPpmFYFla1RA7YyWZQkAOmw6iaZR",
		"ap_ssid": "SmartLife",
		"ap_passwd": null,
		"country_code": "CN",
		"bt_mac": null,
		"bt_hid": null,
		"prod_test": false,
		"fac_pin": "expmpw4xxd0kkifb"
	},
	"gw_di": {
		"abi": 0,
		"id": "ebba4530f41c44cd94bbt6",
		"swv": "1.0.2",
		"bv": "40.00",
		"pv": "2.2",
		"lpv": "3.4",
		"pk": "keyhj9djx7whse3h",
		"firmk": "keyhj9djx7whse3h",
		"cadv": "1.0.5",
		"cdv": "1.0.0",
		"dev_swv": "1.0.2",
		"s_id": "e6hr30",
		"dtp": 0,
		"sync": 0,
		"attr_num": 1,
		"mst_tp_0": 9,
		"mst_ver_0": "1.0.2",
		"mst_md5_0": null,
		"mst_tp_1": 0,
		"mst_ver_1": null,
		"mst_md5_1": null,
		"mst_tp_2": 0,
		"mst_ver_2": null,
		"mst_md5_2": null,
		"mst_tp_3": 0,
		"mst_ver_3": null,
		"mst_md5_3": null
	},
	"tls_ca_cnt": 0,
	"rmt_ctrl_0": {
		"no": 1,
		"mac": "3CNPngKh",
		"grpid": 2,
		"s2": "v7rqfvHr8x9LjtMU4Qvi+g==",
		"sn": 204,
		"cld": 1,
		"key_tp": 0,
		"s1": "pKE="
	},
	"rmt_ctrl_2": {
		"no": 3,
		"mac": "3CNPnIKg",
		"grpid": 2,
		"s2": "b4T6jHldMkB8JxSZlCbC+w==",
		"sn": 212,
		"cld": 1,
		"key_tp": 0,
		"s1": "7aA="
	},
	"gw_wsm": {
		"nc_tp": 9,
		"ssid": null,
		"passwd": null,
		"md": 0,
		"random": 0,
		"wfb64": 1,
		"stat": 2,
		"token": "usKN1mof",
		"region": "AZ",
		"reg_key": "33Mg",
		"dns_prio": 0
	},
	"wf_start_md": 3,
	"rmt_ctrl_1": {
		"no": 2,
		"mac": "3CNPnh4z",
		"grpid": 2,
		"s2": "aeMRFi8ZispIVeATvF2KVA==",
		"sn": 201,
		"cld": 1,
		"key_tp": 0,
		"s1": "HzM="
	},
	"is_stride": 0,
	"ble_beaconkey": "DE564585F7F84349096414B732485157",
	"gw_ai": {
		"key": "y??*6u1H9.SEyf%[",
		"lckey": "~:8:'<:aOayYg|O~",
		"h_url": "http://a.tuyaus.com/d.json",
		"h_ip": "44.241.46.212",
		"hs_url": null,
		"hs_ip": null,
		"hs_psk": "https://a3.tuyaus.com/d.json",
		"hs_psk_ip": "54.69.37.156",
		"mqs_url": null,
		"mqs_ip": null,
		"mq_url": "m2.tuyaus.com:1883",
		"mq_ip": "52.12.196.80",
		"ai_sp": null,
		"ai_sp_ip": null,
		"mq_psk": "m2.tuyaus.com:8886",
		"mq_psk_ip": "52.12.196.80",
		"lp_url": "baal.tuyaus.com",
		"lp_ip": "44.239.29.247",
		"time_z": "-07:00",
		"s_time_z": "[[1710061200,1730620800],[1741510800,1762070400]]",
		"wx_app_id": null,
		"wx_uuid": null,
		"dy_tls_m": 2,
		"cloud_cap": 1025,
		"psk21_key": null
	},
	"e6hr30": [
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 20
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"range": [
					"white",
					"colour",
					"scene",
					"music"
				],
				"type": "enum"
			},
			"id": 21
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 10,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 22
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 23
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 24
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 86400,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 26
		},
		{
			"type": "obj",
			"mode": "wr",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 27
		},
		{
			"mode": "rw",
			"id": 31,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 32,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 33,
			"type": "raw"
		},
		{
			"type": "obj",
			"mode": "ro",
			"property": {
				"min": 1,
				"max": 10000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 46
		},
		{
			"type": "obj",
			"mode": "ro",
			"property": {
				"min": 1,
				"max": 1024,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 47
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 1,
				"max": 1000,
				"scale": 1,
				"step": 1,
				"type": "value"
			},
			"id": 53
		},
		{
			"mode": "rw",
			"id": 56,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 61,
			"type": "raw"
		},
		{
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 71,
			"type": "obj"
		},
		{
			"mode": "rw",
			"id": 210,
			"type": "raw"
		}
	],
	"astro_timer": {
		"timestamp": 1701539485,
		"index": 0
	},
	"mf_test_close": true,
	"em_sys_env": "bk7231n",
	"timer_arr": {
		"lastFetchTime": 0,
		"cnt": 0
	}
}
```

closed 2023-12-02 by Cossid