tuya-cloudcutter/tuya-cloudcutter #551: wlan0: Could not connect to kernel driver

CSSMWX, 2023-12-04:
Trying to flash a tuya KMC-30407 which kindly added to his program.
I'm using a raspberry pi 3B and followed your installation instructions.
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
I use this pi for Tasmota etc.

The first  KMC-30407 is now dead.
 With the second KMC-30407 I added my home WiFi info "-s tasmot 12345678"
sudo ./tuya-cloudcutter.sh -r -s tasmot 12345678 but I cannot get into it.

Please see
https://www.elektroda.com/rtvforum/viewtopic.php?p=20836512#20836512

Cossid, 2023-12-04:
`wlan0: Could not connect to kernel driver` is just a message from the OS, it is not an error.  There is no action to take for it.

If that is the last line you see, it means your device is stuck at the DHCP stage (`-v` would make this more clear).  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

You can't really `kill` a device with cloudcutter, aside from flashing the wrong firmware, which the application has several layers of protection against, you're likely misunderstanding something.

CSSMWX, 2023-12-04:
Thank you for your reply, attached is what I see, there are more in the mentioned thread, inc;uding screenshots etc.
[cloudcutter11.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13551695/cloudcutter11.zip)
I never see cloudcutterflash to select it, cannot enter cloudcutterflash in my phone, there are some "hidden" SSIDs, and when I enter said IDs +pw it get's rejected.
I see  "wlan0" connect mo my AP "tasmot", it receives an IP but alas, I cannot log into  "wlan0".

Also sorry about "dead", it's probably just the pin settings gone wrong, it does not respond to the button, so I cannot put it into AP mode etc.
Thank you for your help with this, we are stuck at this for many days.

Cossid, 2023-12-04:
That log shows a completed cut process.  Cutting leaves the Tuya firmware intact, but cuts it from communicating with the cloud.  `wlan0` is the device name hard-coded into tuya's firmware.  You should be able to use the device with LocalTuya or TinyTuya.  If you're trying to flash 3rd party firmware, that is a different option, and you should not be supplying ssid and password.  If you pass in an ssid and password, it triggers cut mode automatically, since those values aren't needed for 3rd party firmware.

> Also sorry about "dead", it's probably just the pin settings gone wrong, it does not respond to the button, so I cannot put it into AP mode etc.

Sounds like you flashed 3rd part firmware, in which case, it should have presented an AP and you can configure or re-flash it.

CSSMWX, 2023-12-04:
ah, thanks for clearing that wlan0 thing.
If I don't submit the SSID and PW, won't it knock out the unit like the first one?
I did not provide SSID and pw with the first one, and it "plays" dead.
Yes, I'd love to install 
https://github.com/openshwprojects/OpenBK7231T
I am selecting that as you can see.
After your program finishes, I only get the SSID "A-5ADD" from the unit, like before.
However, the subdomain changed from .4. to .175. after a reboot.
And I cannot get to 192.168.4.1 (or 192.168.175.1 after flash/cut reboot)
What do I need to do to get this flashed with OBK please?
Thanks!

Cossid, 2023-12-04:
> If I don't submit the SSID and PW, won't it knock out the unit like the first one?

You'd be installing 3rd party firmware.  It's not "knocking out", that's just how the process works, you have to configure the device to work correctly with your 3rd party firmware.

> I did not provide SSID and pw with the first one, and it "plays" dead.

It's not "playing dead", it has new unconfigured firmware on it.  You need to connect to the AP it is presenting you and configure it.

> After your program finishes, I only get the SSID "A-5ADD" from the unit, like before.
However, the subdomain changed from .4. to .175. after a reboot.
And I cannot get to 192.168.4.1 (or 192.168.175.1 after flash/cut reboot)

If the device still has `A-` prefix, it was only cut and not flashed with 3rd party firmware.  Tuya's firmware does not have a web interface.  If you flash with 3rd party firmware, you must first connect to the AP then either join it to your network, or visit the webpage the AP provides.

> What do I need to do to get this flashed with OBK please?

Read any of the instructions or FAQs, this is all spelled out very clearly, and you're just trying random things expecting it to do exactly what you want, but that's not how anything works.

CSSMWX, 2023-12-04:
That's *exactly* what I did.
I followed the "usage instructions" for a Pi, selected the right unit 
"kmc-30407-smart-tap-mini-4-outlet-plug-bk7231n-v1.1.8.json"
and did use the "custom-firmware" as mentioned there.
This leads to the unit not responding to the switch, setting up an AP, handing out an IP to my laptop and that's it!
No response, no ping, let's call it "zoned out" instead of dead.
Like the first unit, now it's two.
What am I doing wrong? I reread line by line, entered the right choices as you can see.
[cloudcutter12.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13552427/cloudcutter12.zip)

Cossid, 2023-12-04:
Cloudcutter did everything correctly and flashed 3rd party firmware.  Your device now has 3rd party firmware installed, consult their support for further assistance, I've helped all I can here.  CloudCutter provides the facility to install 3rd party firmware, but it doesn't support them.

kmavrov, 2023-12-22:
If i may steal a bit the thread. I have successfully cut the device and now it hangs with an open AP named `A-8404`. My question is how can i force it now to actually connect to my wifi so that i can install some 3rd party firmware? Thanks in advance

Cossid, 2023-12-22:
> If i may steal a bit the thread. I have successfully cut the device and now it hangs with an open AP named `A-8404`. My question is how can i force it now to actually connect to my wifi so that i can install some 3rd party firmware? Thanks in advance

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do