tuya-cloudcutter/tuya-cloudcutter #553: Bullseye 2022-04-04 Lite Install Docker Unsigned or Invalid Repositories

bikerglen, 2023-12-05:
I'm following the Raspberry Pi specific instructions on a RPi 4 4GB model. I'm using the referenced RPi OS version and have followed the instructions to the letter. When I try to build the docker image with
```sudo docker build --network=host -t cloudcutter .```
I get the following errors related to unsigned repos:
```
glen@raspberrypi:~/tuya-cloudcutter $ sudo docker build --network=host -t cloudcutter .
[+] Building 4.0s (6/12)                                                                                                       docker:default
 => [internal] load build definition from Dockerfile                                                                                     0.2s
 => => transferring dockerfile: 462B                                                                                                     0.0s
 => [internal] load .dockerignore                                                                                                        0.2s
 => => transferring context: 2B                                                                                                          0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                           1.1s
 => [internal] load build context                                                                                                        0.3s
 => => transferring context: 3.02kB                                                                                                      0.0s
 => CACHED [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5  0.0s
 => ERROR [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential l  2.2s
------
 > [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto:
1.778 W: GPG error: http://deb.debian.org/debian bullseye InRelease: At least one invalid signature was encountered.
1.778 E: The repository 'http://deb.debian.org/debian bullseye InRelease' is not signed.
1.779 W: GPG error: http://deb.debian.org/debian-security bullseye-security InRelease: At least one invalid signature was encountered.
1.779 E: The repository 'http://deb.debian.org/debian-security bullseye-security InRelease' is not signed.
1.779 W: GPG error: http://deb.debian.org/debian bullseye-updates InRelease: At least one invalid signature was encountered.
1.779 E: The repository 'http://deb.debian.org/debian bullseye-updates InRelease' is not signed.
------
Dockerfile:3
--------------------
   1 |     FROM python:3.9.18-slim-bullseye AS base
   2 |
   3 | >>> RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
   4 |
   5 |     FROM base AS python-deps
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto" did not complete successfully: exit code: 100
```
Is there a fix for this or is there a way to run natively without docker or is there an updated list of instructions for the RPi?
Thanks.

Cossid, 2023-12-05:
Looks entirely like a docker cache issue, I'm using the same version without issue.  Try `docker builder prune` and try again?  See https://stackoverflow.com/questions/62473932/at-least-one-invalid-signature-was-encountered for possible more advanced steps.

bikerglen, 2023-12-05:
same error after running `sudo docker builder prune` looking at the stackoverflow post now.

bikerglen, 2023-12-05:
I modified the Dockerfile to have the allow unauthenticated and allow insecure options: 
```RUN apt-get -qq update && apt-get install -qy --no-install-recommends --allow-unauthenticated --allow-insecure-repositories git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto```
Same error. I tried expanding the RPi file system. I'm on a 16GB micro SD card. Same error.
I'm going to try again in a few hours.

Cossid, 2023-12-05:
Yeah, nothing else I can suggest other than starting fresh.  I'm using the exact same setup as you, RPi4 4GB running 2022-04-04 with the exact same (latest) version of docker, and everything is working as expected.  I think I did have to do some extreme docker cache cleanup when we switched to the most recent docker container, and I couldn't explain why, but since I had no other containers, I could just basically reset docker completely, and no issues after.

bikerglen, 2023-12-05:
Crud. This might be user error. There's 2022-04-04 versions of both buster and bullseye. Retrying with bullseye now.

Cossid, 2023-12-05:
Ah, yeah, I've been using Bullseye.  Wonder if the keys on buster are expired or something.  The error still doesn't make much sense, especially since the command that is failing is in the docker build.

bikerglen, 2023-12-05:
Thanks for the help. That appeared to be the issue. I confused the 2022-04-04 version of buster with the 2022-04-04 of bullseye when searching for the OS image.

closed 2023-12-05 by bikerglen