tuya-cloudcutter/tuya-cloudcutter #554: New Profile Help - Lumary RBBAI String Light, US-SI56B-1 [BK7231N / CBU]

dcode, 2023-12-05:
I don't know exactly what I'm doing, but I'm trying to learn. I cracked open the fused controller case for these lights (15 bulbs) and managed to extract the firmware using my Flipper Zero as a USB-UART bridge. The controller is a BK7231N / CBU. I used `bk7231tools` and it seemingly was successful, but I was unable to make a profile from the bin dump. Hopefully there's enough info here to figure it out.

The device has never been set up or powered on since I got it. I tried cloudcutter with another device, Lumary RGBAI Outdoor Spotlights Pro. They both have the same FCC-ID 2AVB5-SSL and the [internal photos](https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=Y&application_id=E8tXtSAyMGxZklo6kgnssw%3D%3D&fcc_id=2AVB5-SSL) show that the landscape lights also has a BK7231N / CBU. I tested the landscape lights using the test script, which said it was vulnerable. I'm hoping if we can get this profile working, I'll be able to OTA update the landscape lights (which costs more and I wasn't willing to crack it open without knowing what I was doing).

THANKS!

Extracting the flash:
```
$ bk7231tools read_flash -d /dev/ttyACM0 -s 0 -l 0x200000 dump.bin 
Connected! Chip info: 0x7231c / Flash ID: eb 60 15 / Protocol: FULL
Reading 2097152 bytes from 0x0
Reading 4k page at 0x200000 (0.00%)
Reading 4k page at 0x201000 (0.20%)
Reading 4k page at 0x202000 (0.39%)
Reading 4k page at 0x203000 (0.59%)
Reading 4k page at 0x204000 (0.78%)
... truncated ...
Reading 4k page at 0x3FF000 (99.80%)
```

Attempting to build a profile:
```
python3 build_profile.py ../Lumary_RGBAI-String-Light-USSI56B1.bin 
[+] Processing file='../Lumary_RGBAI-String-Light-USSI56B1.bin' as Lumary_RGBAI-String-Light-USSI56B1
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xea20]
		extracted to /home/dcode/Projects/tuya-cloudcutter/Lumary_RGBAI-String-Light-USSI56B1
	0x129f0a: app - INVALID PAYLOAD
Missing app RBL container. Using a scan pattern instead
Traceback (most recent call last):
  File "/home/dcode/Projects/tuya-cloudcutter/profile-building/build_profile.py", line 44, in <module>
    extract.run(file)
  File "/home/dcode/Projects/tuya-cloudcutter/profile-building/extract.py", line 50, in run
    bk7231tools.__main__.dissect_dump_file(input)
  File "/home/dcode/Projects/tuya-cloudcutter/profile-building/.direnv/python-3.11/lib64/python3.11/site-packages/bk7231tools/__main__.py", line 271, in dissect_dump_file
    _, code = __scan_pattern_find_payload(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dcode/Projects/tuya-cloudcutter/profile-building/.direnv/python-3.11/lib64/python3.11/site-packages/bk7231tools/__main__.py", line 191, in __scan_pattern_find_payload
    raise ValueError(
ValueError: First block level CRC-16 checks failed while analyzing partition app
```

I also tried hid_downloader.py, but was never able to successfully read the whole image. Lastly, I tried the C# app BK7231 Easy UART Flasher, it didn't seem to get anywhere either.

K7231 Easy UART Flasher Output:
```
Backup name is set to Lumary_RGBAI-String-Light.
Starting read!
Now is: Tuesday, December 5, 2023 1:40:53 PM.
Flasher mode: BK7231N
Going to open port: /dev/ttyACM0.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 1679 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000200069 data.
Failed! There was no result to save.
```

My extraction setup:
[image]

Focus on board:
[image]

2MB Bin:
[Lumary_RGBAI-String-Light-USSI56B1.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13572109/Lumary_RGBAI-String-Light-USSI56B1.zip)

dcode, 2023-12-05:
Oh, I meant to add in case it eventually helps others, these lights use a WS2814A chip to control the lights. Presumably, the CBU is connected via the SPI pin and not using PWM for color or white temperature control.

Cossid, 2023-12-05:
Nothing you're doing wrong, just this is most likely a newer TuyaOS3 device, which are tools are not fully compatible with, but that would also mean they are patched and serial may be your only option.  This isn't the first image we've run into this with.

It appears TuyaOS3 may have a different partition size for the APP partition.

Cossid, 2023-12-05:
Guess I was wrong, there is indeed something wrong with that dump, it doesn't execute, as the offset of the app partition is wrong.  Can you please make a dump using `ltchiptool`?

dcode, 2023-12-05:
Sure, I'll try it out.

dcode, 2023-12-06:
Okay, ran `ltchiptool` as follows:

```
$ ltchiptool flash read -d /dev/ttyACM0 bk7231n ../Lumary_RGBAI-String-Light-USSI56B1.bin 
I: Connecting to 'Beken 7231N' on /dev/ttyACM0 @ 115200
I: |-- Success! Chip info: 0x7231c
I: Reading Flash (2 MiB) to '../Lumary_RGBAI-String-Light-USSI56B1.bin'
  [################################################################]  100%          
I: |-- Finished in 188.597 s
```

Seemingly, this is the same file I previously dumped, as the sha256 is the same. I also ran it with the generic `bk72xx` family name, same hash result.

```
sha256sum Lumary_RGBAI-String-Light-USSI56B1.bin | tee SHA256SUM
8ad131a7b03ad955fbe546d51f27f006609830f39f38ae152140ed2cb69aa85f  Lumary_RGBAI-String-Light-USSI56B1.bin
```

Added the SHA256SUM to the zip: [Lumary_RGBAI-String-Light-USSI56B1.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13583480/Lumary_RGBAI-String-Light-USSI56B1.zip)

---
One thing that seems odd to me, but I'm not sure, when I run the dump, the board doesn't make me reset it by grounding CEN or disconnecting power after the first time. I assume that's because the baud is already negotiated, but unsure. If I add this device to my Tuya Cloud dev account, will there be any additional information that can help unlock this?

Cossid, 2023-12-06:
This "new" file has the exact timestamp as the previous one, something isn't right.

Cossid, 2023-12-06:
> One thing that seems odd to me, but I'm not sure, when I run the dump, the board doesn't make me reset it by grounding CEN or disconnecting power after the first time.

This is starting to make more sense.  This is because the firmware as it is on the device is corrupted, and your device is already in a boot loop.

Cossid, 2023-12-06:
To further confirm, you could hook up to TX2 and log the serial output as you turn the device on.  I'm assuming you'll see a loop of bootloader text failing to load 0x10000

dcode, 2023-12-06:
> I assume you haven't flashed ESPHome to the device yet? Does it still work, i.e. does the stock firmware still run?
> 
> Could you also attach a ROM dump made with ltchiptool? (IIRC `-R` parameter should make it dump ROM)

Doesn't look like that's an option for this family.

```
ltchiptool flash read --rom -d /dev/ttyACM0 bk7231n ../ltchiptool_bk72xx_rom_20231206T14:48Z.bin 
I: Connecting to 'Beken 7231N' on /dev/ttyACM0 @ 115200
I: |-- Success! Chip info: 0x7231c
E: NotImplementedError: Memory type not readable via UART
E: |-- File "/home/dcode/Projects/tuya-cloudcutter/profile-building/.direnv/python-3.11/lib64/python3.11/site-packages/ltchiptool/soc/bk72xx/flash.py", line 171, in flash_get_size
```

I was able to read from TX2. I get a loop of this, looping through several different IRQs.

```
V:BK7231N_1.0.1
REG:cpsr     spsr     r13      r14
SVC:000000D3          00401C1C 000033AC
IRQ:000000d2 00000010 00401e0c 00005320 
FIR:000000d1 00000010 00401ffc f0fc9078 
SYS:000000df          0040192c 00000158 
ST:0001AD12
J 0x10000
[ARM ANOMALY][2]
```

So... seemingly I bricked it? This is why I got a somewhat cheaper device to test on. Is it possible there's insufficient power? I read that the 3.3V from the Flipper Zero is pretty solid. I had to order a new multimeter, so I'll check that when it arrives. Plugging the controller in, it does seem dead. Last thing, is it possible to just write the new firmware and reverse engineer the pinout assignments?

On the upside, now I can use this to learn and practice chip desoldering to replace with an ESP32 I guess, or maybe just [replace the controller](https://bikerglen.com/blog/controlling-inexpensive-led-lights-with-wled/) altogether.

Cossid, 2023-12-06:
@dcode following up, it appears likely at some point you used the `test write pattern` or `test read/write` functions in BK7231 Easy Flasher, and this permanently corrupts the flash.  Do not use that option without doing a full backup in the future.