tuya-cloudcutter/tuya-cloudcutter #558: Add new device

espilioto, 2023-12-07:
Here is the lightleak dump:
[Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13603235/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb.zip)

Tried building a profile but this was the output:
```
soul@mintvm:~/Desktop/tuya-cloudcutter/profile-building$ python3 build_profile.py ~/Desktop/2023-12-07_19-14-48_lightleak/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb.bin 
[+] Processing file='/home/soul/Desktop/2023-12-07_19-14-48_lightleak/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb.bin' as Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xea20]
		extracted to /home/soul/Desktop/2023-12-07_19-14-48_lightleak/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb
	0x129f0a: app - [encoding_algorithm=NONE, size=0xf4160]
		extracted to /home/soul/Desktop/2023-12-07_19-14-48_lightleak/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb
	0x210f9a: bootloader - INVALID PAYLOAD
Storage partition:
!!! Couldn't unpack KVStorage, perhaps dump file is invalid?
Traceback (most recent call last):
  File "/home/soul/.local/lib/python3.10/site-packages/bk7231tools/__main__.py", line 307, in dissect_dump_file
    kvs = KVStorage.unpack(storage_data, aes=aes)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 455, in unpack
    raise e
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 446, in unpack
    value = cls._read_field(ctx, field, meta)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 327, in _read_field
    item = cls._read_field(ctx, base_field, base_meta)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 284, in _read_field
    adapted = cls._read_value(ctx, meta, field.type)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 261, in _read_value
    return typ.unpack(io=ctx.G.io, parent=ctx, **kwargs)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 455, in unpack
    raise e
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 446, in unpack
    value = cls._read_field(ctx, field, meta)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/main.py", line 302, in _read_field
    return evaluate(ctx, meta.action)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/utils/context.py", line 13, in evaluate
    value = v(ctx)
  File "/home/soul/.local/lib/python3.10/site-packages/datastruct/fields/helper.py", line 147, in _validate
    raise ValueError(f"Validation failed at '{doc}'")
ValueError: Validation failed at 'block magic'; while unpacking 'DataBlock._magic'; while unpacking 'KVStorage.data_blocks'
App code `user_param_key`:
	- found! Extracted to /home/soul/Desktop/2023-12-07_19-14-48_lightleak/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb/Eurolamp_SmartWifiPlus-GU10-6w-RGBCT-Bulb_user_param_key.json
[+] Searching for known exploit patterns
[+] Matched pattern for BK7231N version SDK 2.3.1, payload type ssid
[+] Searching for ssid payload address
[+] ssid payload address gadget (THUMB): 0xD3ABD
[+] Searching for finish address
[+] Finish address gadget (THUMB): 0x1B39B
[!] Storage file not found
[+] SDK: 2.3.1
[+] Device class: oem_bk7231n_light_ty
[+] Version: 1.2.16
[+] bv: 40.00
[+] firmware_key: key8u54q9dtru5jw
[!] uuid was not found
[!] Unable to build complete profile as schema remains missing.
```

Did I miss something in lightleak?
Duped it twice in case I forgot to tap the storage button.

Cossid, 2023-12-07:
Nothing you did wrong, just a bug in lightleak where if you press both read flash and read storage, it uses the same file, and it's technically non-standard.  Just needs a bit of cleanup (truncating at 0x200000).  I'll add a new profile shortly.

espilioto, 2023-12-07:
Love you dude <3
Will also post a teardown in a bit!

Cossid, 2023-12-07:
Added as `EUROLAMP 147-77903 URGCW1.3 6W GU10 RGBCT Bulb`

closed 2023-12-07 by Cossid