tuya-cloudcutter/tuya-cloudcutter #559: Docker build hangs at "Installing dependencies from Pipfile.lock (019fbf)"

ilium007, 2023-12-08:
My build just hangs here all the time. 

```
xxx@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

```
xxxx@raspberrypi:~/tuya-cloudcutter $ sudo docker build --network=host -t cloudcutter .
[+] Building 131.1s (9/12)                                                                                                                       docker:default
 => [internal] load .dockerignore                                                                                                                          0.2s
 => => transferring context: 2B                                                                                                                            0.0s
 => [internal] load build definition from Dockerfile                                                                                                       0.2s
 => => transferring dockerfile: 462B                                                                                                                       0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                                             3.8s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5505ae04f1                 16.5s
 => => resolve docker.io/library/python:3.9.18-slim-bullseye@sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5505ae04f1                       0.1s
 => => sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5505ae04f1 1.86kB / 1.86kB                                                             0.0s
 => => sha256:5c341207bb00b5bcef01cd71e52a2b1741fd811ce36f966dc3931f7d492ea655 1.37kB / 1.37kB                                                             0.0s
 => => sha256:dba4c9065a53d8cf3c80f5679e5eccc134eb6694e02df6030deabd3fa37b44bd 6.91kB / 6.91kB                                                             0.0s
 => => sha256:6dc4ed5513793308b8e30b08e03f97fa54025c3d3f3172c6edccb1dbbc7ff139 26.58MB / 26.58MB                                                           7.5s
 => => sha256:41fb338b21e7a0e0e5a6b6c648c9329d491cce47f9628c490a41bed4006e9f29 10.38MB / 10.38MB                                                           4.0s
 => => sha256:77bbb4df0219ace7c34407004596f50c37aa4245fbf341b00147a6ddbd4e4417 837.53kB / 837.53kB                                                         0.8s
 => => sha256:a74f2f4eedf621478d8467803e04ad5458b9658c979a4933fa0ed2bc6fa73c42 233B / 233B                                                                 1.1s
 => => sha256:c6e9687277c346128a7b34f9f41e1872219768a06feb9a876619d700e17a9547 2.93MB / 2.93MB                                                             2.6s
 => => extracting sha256:6dc4ed5513793308b8e30b08e03f97fa54025c3d3f3172c6edccb1dbbc7ff139                                                                  4.9s
 => => extracting sha256:77bbb4df0219ace7c34407004596f50c37aa4245fbf341b00147a6ddbd4e4417                                                                  0.3s
 => => extracting sha256:41fb338b21e7a0e0e5a6b6c648c9329d491cce47f9628c490a41bed4006e9f29                                                                  1.7s
 => => extracting sha256:a74f2f4eedf621478d8467803e04ad5458b9658c979a4933fa0ed2bc6fa73c42                                                                  0.0s
 => => extracting sha256:c6e9687277c346128a7b34f9f41e1872219768a06feb9a876619d700e17a9547                                                                  0.9s
 => [internal] load build context                                                                                                                          0.2s
 => => transferring context: 855.04kB                                                                                                                      0.1s
 => [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosq  66.9s
 => [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                    24.8s
 => [python-deps 2/4] COPY src/Pipfile /src/                                                                                                               0.2s
 => [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                          0.2s
 => [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                     18.4s
 => => #   seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
 => => #     added seed packages: pip==23.3.1, setuptools==69.0.2, wheel==0.42.0
 => => #   activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
 => => # ✔ Successfully created virtual environment!
 => => # Virtualenv location: /src/.venv
 => => # Installing dependencies from Pipfile.lock (019fbf)...
```

ilium007, 2023-12-08:
over 10mins so far

`[+] Building 614.1s (9/12)`

ilium007, 2023-12-08:
All good - just didn't wait long enough. Working now... Well, not working but I'll raise a seperate issue.

closed 2023-12-08 by ilium007

aletzi1, 2023-12-08:
Ill have similar issue, been waiting over 30min and still going.

aletzi1, 2023-12-09:
> > Ill have similar issue, been waiting over 30min and still going.
> 
> It just takes time, depending on the platform. It's about 15 minutes on an RPi4, I've heard 30-45 minutes on older RPis. I've looked into changing the dependency validator, including poetry and pdm, but nothing made a significant difference.

Thanks for reply. Took over 4000 seconds (1hour) with raspberry pi 3b to get over "Installing dependencies from Pipfile.lock (019fbf)..."
But now itse working. 😁

Cossid, 2023-12-13:
Found the dependency issue on an upstream tool and got it fixed, build times should be considerably shorter now.

On a RPi4, it went from about 15 minutes to about 3 minutes for a full clean docker build.