tuya-cloudcutter/tuya-cloudcutter #561: I have messed up - ran cloudcutter with wrong device id. Now can't re-run due to keys

ilium007, 2023-12-08:
I have an Arlec PCA191 Series 2. I mistakingly ran the 1.3.5 patch instead of the 1.1.8 and also didn't realise I had to select the esphome firmware.

I am trying to re-run but keep getting this:

```
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying is false - bailing out
```

The device is currently connected to my wifi network (ran sudo ./tuya-cloudcutter.sh -s xxx xxxx initially)

I figured I could go back and find the authkey, uuid and pskkey from the "configued-devices" file from the first run. This didn't work:

```
xxxx@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -r -v -a "aCF3tmQ6kpHyMltq" -u "9n3IX9dGj2P9" -k "JCDsJG3AxDkDQVWV"
Building cloudcutter docker image
[+] Building 1.2s (13/13) FINISHED                                                                                                               docker:default
 => [internal] load build definition from Dockerfile                                                                                                       0.1s
 => => transferring dockerfile: 462B                                                                                                                       0.0s
 => [internal] load .dockerignore                                                                                                                          0.0s
 => => transferring context: 2B                                                                                                                            0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                                             0.9s
 => [internal] load build context                                                                                                                          0.0s
 => => transferring context: 3.02kB                                                                                                                        0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9874a82704a8920109d161deb1f8560666945787fa9ee046e00caf5505ae04f1                  0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                              0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                        0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                   0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                               0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                 0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                  0.0s
 => exporting to image                                                                                                                                     0.0s
 => => exporting layers                                                                                                                                    0.0s
 => => writing image sha256:8d0782000138dd74f1c46faa023537f3e22779a56abb5dd892071c210f6f45e8                                                               0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                             0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Arlec
   Alantop
   Aldi
   Alic
   Amico
   Antela
   Aoycocr
 > Arlec
   Atarm
   Athom
   Atomi
   Aubess
   Avatar
   Avatto

[?] Select the article number of your device: PC191HA Smart Plug BK7231N v1.1.8
   ALD092CHA CCT Downlight
   ALD095HA RGBCT Downlight
   ALP223HA Desktop Bar Light Twin Pack
   GLD12x HA Series 2 RGB E27 Bulb
   GLD130HA
   Grid Connect GLD132HA Bulb
 > PC191HA Smart Plug BK7231N v1.1.8
   PC191HA Smart Plug BK7231T v1.3.5
   PC193HA Smart Plug With USB
   RGB B22 Bulb

[?] Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug
 > 1.1.8 - BK7231N / oem_bk7231n_plug

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231n.ug.bin

Selected Device Slug: arlec-pc191ha-smart-plug-bk7231n-v1.1.8
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Using AuthKey aCF3tmQ6kpHyMltq , UUID 9n3IX9dGj2P9 , and PSKKey JCDsJG3AxDkDQVWV
Writing deviceconfig file...
usage: cloudcutter write_deviceconfig
       [-h]
       [--output-directory OUTPUT_DIRECTORY]
       [--deviceid DEVICE_ID]
       [--localkey LOCAL_KEY]
       --authkey
       AUTH_KEY
       --uuid
       UUID
       --pskkey
       PSK_KEY
       profile
       verbose_output
cloudcutter write_deviceconfig: error: argument --authkey: invalid check_arg value: 'aCF3tmQ6kpHyMltq'
Oh no, something went wrong with running the exploit! Try again I guess...
```

Contents of configured-devices/9n3IX9dGj2P9.deviceconfig (this was the initial successful cloudcutter run)
`
{"uuid": "9n3IX9dGj2P9", "auth_key": "aCF3tmQ6kpHyMltq", "local_key": "JCDsJG3AxDkDQVWV", "sec_key": "diMVubWuN68O4VWG", "device_id": "LWvjahLcpv7oEg7NxBBy", "chip_family": "BK7231T", "profile_name": "1.3.5 - BK7231T / oem_bk7231s_dltj", "device_slug": "arlec-pc191ha-smart-plug-bk7231t-v1.3.5"}`

Cossid, 2023-12-08:
Don't run with `-a` or `-u`, those are only for very specific cases.  You originally attempted an exploit, and those were the values it tried to set and failed, they aren't valid values.  For the regular process, it will generate new ones each time, and if successful use the generated ones.  `-a` and `-u` are for skipping the exploit step, which you don't want or need to do in this case.

closed 2023-12-08 by Cossid

Cossid, 2023-12-08:
No, it is not bricked, just run it again without those parameters.

Cossid, 2023-12-08:
To clarify, it sounds like you ran the 1.3.5 for BK7231T and it worked.  1.3.5 on T is *not* patched.  1.3.5 on N *is* patched.

It sounds like 1.3.5 T is what your device actually is.

ilium007, 2023-12-08:
I will try the 1.3.5 ESPHome firmware flash again...

ilium007, 2023-12-08:
That worked!