tuya-cloudcutter/tuya-cloudcutter #562: Mini Universal IR Remote S18 CB3S BK7231n

Vladnwx, 2023-12-09:
{
	"reset_pin":"6",
	"wfst_pin":"8",
	"owm":"1",
	"reset_lv":"0",
	"rsthold":"3",
	"wfst_lv":"1",
	"netyc":"1",
	"infrr":"7",
	"netnc":"0",
	"infre":"26",
	"crc":"42",
	"}ac_pin":"ejv37o6yz9vtxcuobJ0Agw_di{abi",
	"id":"null",
	"swv":"2.0.0",
	"bv":"40.00",
	"pv":"2.2",
	"lpv":"3.3",
	"pk":"key54vrth5askhsj",
	"firmk":"key54vrth5askhsj",
	"cadv":"1.0.3",
	"cdv":"1.0.0",
	"dev_swv":"2.0.0",
	"s_id":"null",
	"dtp":"0",
	"sync":"0",
	"attr_num":"0",
	"mst_tp_0":"0",
	"mst_ver_0":"null",
	"mst_tp_1":"0",
	"mst_ver_1":"null",
	"mst_tp_2":"0",
	"mst_ver_2":"null",
	"mst_tp_3":"0",
	"mst_ver_3":"null }{nc_tp",
	"ssid":"null",
	"passwd":"null",
	"md":"0",
	"random":"0",
	"wfb64":"1",
	"stat":"0",
	"token":"null",
	"region":"null",
	"reg_key":"null",
	"dns_prio":"0 } )A0Agw_wsm{nc_tp",
	"psk_key":"2N7hBsaHXQmjk2FT9UlKXmvWg7FLSJIbwenZ3",
	"auth_key":"8SDaHdYzoa4AgD3bfgXez2dUxD9desdi",
	"ap_ssid":"SmartLife",
	"ap_passwd":"null",
	"country_code":"null",
	"bt_mac":"null",
	"bt_hid":"null",
	"prod_test":"false",
	"fac_pin":"ejv37o6yz9vtxcuosK-Agw_ai{key",
	"lckey":"null",
	"h_url":"null",
	"h_ip":"null",
	"hs_url":"null",
	"hs_ip":"null",
	"hs_psk":"null",
	"hs_psk_ip":"null",
	"mqs_url":"null",
	"mqs_ip":"null",
	"mq_url":"null",
	"mq_ip":"null",
	"ai_sp":"null",
	"ai_sp_ip":"null",
	"mq_psk":"null",
	"mq_psk_ip":"null",
	"time_z":"null",
	"s_time_z":"null",
	"w{uuid":"d8140ead423dc968",
	"wx_uuid":"null",
	"dy_tls_m":"0",
	"cloud_cap":"0",
	"psk21_key":"null } )A0Agw_wsm{nc_tp",
	"ap_s{reset_pin":"6",
	"Agw_wsm{nc_tp":"0"
}

Sorry, no meaningful pins data found. This device may be TuyaMCU or a custom one with no Tuya config data.
No module information found.
And the Tuya section starts, as usual, at 2023424

kuba2k2, 2023-12-09:
IR transmitter on pin P26.
IR receiver on pin P7.
Reset button on pin P6.
Status LED on pin P8.

The data returned by "BK7231 Easy UART Flasher" is usually incomplete and has malformed values.

Cossid, 2023-12-09:
Yeah, this is all the values in the key/value section, but not separated by sections and with corruption between.  We can't easily do anything with this directly.

closed 2023-12-09 by Cossid