tuya-cloudcutter/tuya-cloudcutter #563: Hombli HBSS 0109 Exploit fails

Xploder, 2023-12-11:
Hey, I'm having trouble with cloudcutter and Hombli HBSS 0109 @ version 1.3.5 (there's a profile for it)

[image]

Been trying this on two host devices, double checked for version 1.3.5 in SmartLife before removal. Chip is WB2S.

Any tips?

Cossid, 2023-12-11:
That profile was based off the information provided in https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/564

That is the only profile available for 1.3.5.  It's possible they have a different firmware with the same version, but we would need a firmware dump to verify, otherwise I can only recommend trying again.  Being that you verified it is WB2S (BK7231T), we can rule out it isn't the 1.3.5 for BK7231N which is patched.

Cossid, 2023-12-22:
Closing as stale, as there isn't anything we can do without further information.  Reopen if you have more information to provide.

closed 2023-12-22 by Cossid