tuya-cloudcutter/tuya-cloudcutter #568: Support for new device TYUA PM2.5 HUA6-3P, WR3 chip inside

alexreddy78, 2023-12-11:
Dear Developer,

Pls kindly check and advise if possible to add support for flw data:

    "gw_di": {
        "abi": 0,
        "id": "bf820b121719650882hvkl",
        "swv": "1.0.5",
        "bv": "30.07",
        "pv": "2.2",
        "lpv": "3.3",
        "pk": "75erf4zsyfldtzp2",
        "firmk": "keym5sjxrqcryptn",
        "cadv": "1.0.2",
        "cdv": "1.0.0",
        "dev_swv": "1.0.3",
        "s_id": "00000209zb",
        "dtp": 1,
        "sync": 0,
        "attr_num": 0,
        "mst_tp_0": 0,
        "mst_ver_0": null,
        "mst_tp_1": 0,
        "mst_ver_1": null,
        "mst_tp_2": 0,
        "mst_ver_2": null,
        "mst_tp_3": 0,
        "mst_ver_3": null,
        "dminfo_name": null,
        "dminfo_code": null
    },
    "gw_wsm": {
        "nc_tp": 3,
        "ssid": "n/a",
        "passwd": "n/a",
        "md": 0,
        "random": 0,
        "wfb64": 1,
        "stat": 2,
        "token": "BArKuAF8",
        "region": "EU",
        "reg_key": "9UTa"
    },
    "tls_ca_cnt": 0,
    "gw_ai": {
        "key": "n/a",
        "lckey": "n/a",
        "h_url": "http://a.tuyaeu.com/d.json",
        "hs_url": null,
        "hs_psk": "https://a3.tuyaeu.com/d.json",
        "mqs_url": null,
        "mq_url": "m2.tuyaeu.com:1883",
        "ai_sp": null,
        "mq_psk": "m2.tuyaeu.com:8886",
        "time_z": "+03:00",
        "s_time_z": "[]",
        "wx_app_id": null,
        "wx_uuid": null,
        "dy_tls_m": 0
    },
    "timer_arr": {
        "lastFetchTime": 0,
        "cnt": 0
    },
    "00000209zb": [
        {
            "mode": "ro",
            "property": {
                "min": 0,
                "max": 999,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "trigger": "direct",
            "id": 102,
            "type": "obj"
        },
        {
            "mode": "ro",
            "property": {
                "min": -9,
                "max": 100,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "trigger": "direct",
            "id": 105,
            "type": "obj"
        },
        {
            "mode": "ro",
            "property": {
                "min": 0,
                "max": 99,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "trigger": "direct",
            "id": 106,
            "type": "obj"
        },
        {
            "mode": "ro",
            "property": {
                "range": [
                    "FINE",
                    "GOOD",
                    "OK",
                    "BAD"
                ],
                "type": "enum"
            },
            "id": 107,
            "type": "obj"
        }
    ],
    "em_sys_env": "",
    "ap_info": {
        "ssid": "n/a",
        "passwd": "n/a",
        "chan": 9,
        "security": 4194308
    }
}

Do you need something else except above mentioned?

BRGDS

Cossid, 2025-12-09:
We now support vulnerable RTL8710BN devices, however, we still generally need a dump to support most devices.

Cossid, 2025-12-09:
TuyaMCU firmware 1.0.5 is unfortunately among those not known to be vulnerable https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware#rtl8710bn but a new dump would help with verifying that.